Klipper3d/klipper #160: Move queue empty during virtual SD print

jakep82, 2018-02-03:
I made several successful prints from SD, but this one stopped mid-print with the error below.  Currently running the virtual SD branch.  Log attached.

Recv: // MCU 'mcu' shutdown: Move queue empty
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: !! Printer is shutdown

[klippy.log](https://github.com/KevinOConnor/klipper/files/1692285/klippy.log)

KevinOConnor, 2018-02-04:
On Sat, Feb 03, 2018 at 11:02:10PM +0000, jakep82 wrote:
> I made several successful prints from SD, but this one stopped mid-print with the error below.  Currently running the virtual SD branch.  Log attached.
>
> Recv: // MCU 'mcu' shutdown: Move queue empty [...]

This doesn't appear to be related to the SD code.  Instead, it looks
like there was a large amount of jitter between the system clock and
the PRU clock (they do have different clock circuits).

It's reported in the log as a bunch of "Resetting prediction variance"
messages - the code is complaining that it can't make a good clock
prediction.  This is the first time I've seen this come up.

Was there anything in the Linux system logs around the time of the
event?  Perhaps NTP messages?

I have some ideas on how we could make the code more resilient to this
type of jitter, but I'll need to think about it a bit.

-Kevin

jakep82, 2018-02-04:
The only thing I'm seeing is a little spam from rsyslogd-2007.  This is repeated at random intervals from 30 seconds to 20+ minutes.

Feb  3 21:32:27 beaglebone rsyslogd-2007: action 'action 17' suspended, next retry is Sat Feb  3 21:32:57 2018 [try http://www.rsyslog.com/e/2007 ]

cmock, 2018-02-04:
That rsyslog message is not time related, see [here](https://blog.dantup.com/2016/04/removing-rsyslog-spam-on-raspberry-pi-raspbian-jessie/).

Are you running ntpd on your raspberry?

jakep82, 2018-02-04:
I'm not knowledgeable enough to say for sure.  It's a fresh installation of the most recent Jessie IoT on a BeagleBone Black Wireless.  All I've installed is Klipper, Octoprint, and mjpg-streamer.

jakep82, 2018-02-05:
I've made about half a dozen successful prints since this occurred with zero issues. I don't know what happened, and I've not been able to reproduce it.

KevinOConnor, 2018-02-05:
On Sun, Feb 04, 2018 at 04:52:02PM -0800, jakep82 wrote:
> I've made about half a dozen successful prints since this occurred with zero issues. I don't know what happened, and I've not been able to reproduce it. 

Yeah, I'm not surprised it isn't reproducible, as it's a pretty
unusual timing issue (the host hardware clock drifts significantly
from the mcu hardware clock over a short time period).

FYI, I've pushed commit 08874b9c in response to the original issue.

-Kevin

jakep82, 2018-02-05:
I'm going to go ahead and close this issue.  It appears to be a rare edge case that I can't reproduce, and you've likely addressed it in that commit.

closed 2018-02-05 by jakep82