Klipper3d/klipper #161: Syntax error on Multi MCU with reverse polarity (!)

megablue, 2018-02-05:
this is the part of the config file that triggered an error

`[extruder]
step_pin: extramcu:ar26
dir_pin: extramcu:ar28
enable_pin: extramcu: !ar24
`

Syntax error trigged on enable_pin with the ! sign 
The config works as soon as the ! sign is removed however the EN pin polarity is reversed.

> Recv: // Unable to translate pin name: config_digital_out oid=1 pin=!ar24 value=0 default_value=0 max_duration=0
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: !! Printer is halted

TunaLatte, 2018-02-05:
Always post klipper.log, it will also show the config and targets.

KevinOConnor, 2018-02-05:
On Mon, Feb 05, 2018 at 06:57:54AM +0000, Evert Chin wrote:
> this is the part of the config file that triggered an error
> 
> `[extruder]
> step_pin: extramcu:ar26
> dir_pin: extramcu:ar28
> enable_pin: extramcu: !ar24
> `

It should be:

enable_pin: !extramcu:ar24

-Kevin

megablue, 2018-02-05:
alright, thanks!

closed 2018-02-05 by megablue