Klipper3d/klipper #164: Automaticly issue firmware_restart on reconnect

psy0rz, 2018-02-08:
My printer will completely shutdown when a print is done. (i'm using the PSU plugin for octoprint)

However, when i power the printer on remotely to print something, i always have to manually issue a firmware_restart.

Can we get a flag to enable automatic firmware restart on reconnect? 

If you can point me to the correct file/function i can make a PR for it.

Edwin

TunaLatte, 2018-02-08:
having it manual is beneficial when something goes wrong i guess?
what does adding "M80; firmware_restart" to _PSU on command_ do in PSU control plugin?

psy0rz, 2018-02-08:
i'm using system commands instead of GCODE.

psy0rz, 2018-02-08:
i also have a plugin that automaticly reconnects to "serial" (klippy), and on connect i issue a firmware_restart. 

that works, but offcourse all the commands directly after firmware_restart are ignored.

KevinOConnor, 2018-02-10:
Please provide a klippy.log file of a session containing the disconnect (shutdown) and then the reconnect (followed by firmware_restart).

psy0rz, 2018-02-11:
I removed the log, restarted klippy and then powered the printer. Everything went fine. After i powercycled the printer, a firmware_restart was needed to get everything working again. (i did issue the commands manually instead of letting octoprint do it on connect this time)
[klippy.log](https://github.com/KevinOConnor/klipper/files/1714681/klippy.log)

KevinOConnor, 2018-02-11:
On Sun, Feb 11, 2018 at 10:23:43AM -0800, DatuX wrote:
> I removed the log, restarted klippy and then powered the printer. Everything went fine. After i powercycled the printer, a firmware_restart was needed to get everything working again. (i did issue the commands manually instead of letting octoprint do it on connect this time)

Okay.  If I understand correctly, you have some external mechanism to
remove power to the micro-controller, and you are using that when the
printer is idle.

As you've figured out, the Klipper host code doesn't like that.  To
Klipper, the micro-controller software and host software are
conceptually a "single firmware" - so it doesn't really make sense to
externally remove power to "half the printer".

If there is interest in extending the code to better support this, I
would recommend enhancing Klipper so that it knows how to remove power
to the micro-controller on idle.  Then, it can also be enhanced to
reconnect to the micro-controller when the printer is no longer idle.

I would not recommend implementing an auto-reconnect mechanism.  Such a
mechanism could result in poor error handling should a hardware
failure result in an unexpected communication outage between host and
micro-controller.

-Kevin

psy0rz, 2018-02-12:
You are correct.

Maybe i should adjust octopi or the PSU plugin so that it can pause for the firmware to restart?

Does klippy send some kind of event of ACK when the restart is done? (didnt check this myself yet)

lenne0815, 2018-02-24:
@KevinOConnor Im seeing a very similar issue on my rumba, everytime i power it up i have to reconnect / restart firmware until it picks up communication again. In my case i can probably fix it by setting the rumba to power the atmega via the raspberrys usb.

closed 2018-03-13 by KevinOConnor