Klipper3d/klipper #166: Probe with offset

psy0rz, 2018-02-10:
My probe has a X and Y offset relative to the head printhead. 

When my X and Y axes are homed (its in the corner on my setup), the probe is outside the bed, so when it probes it wont see the bed and crash the head into the build plate :(

I'm trying to figure out how to add a probeoffset to the probe code.

Also: Do the x,y coordinates where the  z-home probe take place have any influence on the bed leveling code? E.g.: Is it just ok to do the z-home probe anywhere? (I prefer to do my initial probe in the middle of the bed)

KevinOConnor, 2018-02-10:
It's not clear what you attempted.  Please attach a klippy.log file of a session (so that we can see your config and any errors raised).  If you use the probe to home the Z, then you can use the homing_override directive to force the xy position prior to the z homing.  If you're using bel_tilt_calibrate, then you specify the probe points as part of the bed_tilt_calibrate configuration.

psy0rz, 2018-02-11:
What i'm trying to do when G28 is issued:

- Home X and Y
- Move X and Y to X197 Y160
- Home Z via probe.

klippy.log when trying that: 
```
Stats 2199529.7: gcodein=122 mcu: mcu_awake=0.006 mcu_task_avg=0.000069 mcu_task_stddev=0.000058 bytes_write=2476 bytes_read=15388 bytes_retransmit=9 bytes_invalid=0 send_seq=211 receive_seq=211 retransmit_seq=2 srtt=0.011 rttvar=0.005 rto=0.033 ready_bytes=0 stalled_bytes=0 freq=16000432 print_time=122.308 buffer_time=0.000 print_stall=0 extruder: target=0 temp=15.2 pwm=0.000 heater_bed: target=0 temp=13.9 pwm=0.000
Stats 2199530.7: gcodein=122 mcu: mcu_awake=0.006 mcu_task_avg=0.000069 mcu_task_stddev=0.000058 bytes_write=2893 bytes_read=15592 bytes_retransmit=9 bytes_invalid=0 send_seq=223 receive_seq=223 retransmit_seq=2 srtt=0.009 rttvar=0.006 rto=0.032 ready_bytes=0 stalled_bytes=0 freq=16000432 print_time=123.424 buffer_time=0.078 print_stall=0 extruder: target=0 temp=14.8 pwm=0.000 heater_bed: target=0 temp=13.2 pwm=0.000
Must home axis first: 197.000 160.000 0.650 [0.000]
Must home axis first: 197.000 160.000 0.650 [0.000]
```

printer.cfg file: https://gist.github.com/psy0rz/7bb2dc4e0bc3316b2b0b111d0c6210ca

psy0rz, 2018-02-11:
ok, when i disable the bed-tilt section everything is fine, revised config: https://gist.github.com/psy0rz/a3617de4218a183f58d7e3e6f3e3bf00

If i comment out all the bed-tilt stuff, then G28 works like expected.

If I do enable bed-tilt, i get the error mentioned in the previous comment.

KevinOConnor, 2018-02-11:
> ok, when i disable the bed-tilt section everything is fine, revised config

In the future, please attach the klipper log file as described at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

What you are likely seeing is that the "G0 X197 Y160 F6000" command in your homing_override section requires Z axis movement (the bed_tilt code configures gcode xy moves to have a small z lift).  However, you have not yet homed the Z axis.  I would try setting "set_position_z = 5" and make sure to run "G28 Z0" at the end of the homing_override gcode.  This will force the printer to believe the head is at Z=5 at the start of the homing, which should then permit the "G0 X197 Y160" command to be executed.

psy0rz, 2018-02-11:
cool that works!

I wondered where set_position_.. was for. The comment says:
```
#   If specified, the position to set the printer to (in mm, relative
#   to the coordinate system specified in the config file) prior to
#   running the above g-code commands. Setting this may be useful if
#   an axis must move prior to invoking the normal G28 mechanism for
#   that axis. The default is to not force a position for the axis.
```

Perhaps that comment should say: "...assume the axis is in this position..." ?

KevinOConnor, 2018-02-11:
On Sun, Feb 11, 2018 at 06:04:56PM +0000, DatuX wrote:
> Perhaps that comment should say: "...assume the axis is in this position..." ?

Makes sense.  I updated the comment.

-Kevin

psy0rz, 2018-02-12:
Awesome

closed 2018-02-12 by psy0rz

andremariz, 2018-11-02:
I have the same problem but i could work it out yet. 
Psy0rz did you make it work? If so, could you share your config so i try to adapt to my ender 3.
This is just for homing;
 And doing the probing to level the bed? Do i use G29 as i used in marlin?

psy0rz, 2018-11-08:
i did not test any further, i have a prusa mk3 now.

andremariz, 2018-11-08:
I got it working. Thx