Klipper3d/klipper #169: BLTouch support

jonathanlundstrom, 2018-02-11:
Hi @KevinOConnor!

I'm really intrigued by this project and I'd love to try it out. Only software feature I'm missing is mesh leveling for the bed which I understand has not yet been implemented.

I learned that bed tilt compensation was recently added but I'm wondering about BLTouch support. As you might know this does not behave like a regular sensor since it's servo based and needs proper commands to deploy/stow.

Perhaps code from Marlin could be borrowed for both BLTouch and mesh leveling, assuming it's then translated to Python instead of C.

Thanks!

Regards,
Jonathan

KevinOConnor, 2018-02-11:
Other users have reported success with the BLTouch.  I added a config file sample to demonstrate how it can be used ( https://github.com/KevinOConnor/klipper/blob/master/config/sample-bltouch.cfg ).

There has been interest expressed in mesh bed leveling (see issue #96).  I don't have any short term plans to work on it, but am open to merging it if someone wishes to code it.

-Kevin

jonathanlundstrom, 2018-02-12:
Awesome, I need to look into this. Does bed tilt compensation work using the G29 command, and if so, does it work combined with BLTouch probing?

closed 2018-02-25 by KevinOConnor