Klipper3d/klipper #171: Thermocouple support.

rewolff, 2018-02-13:
My "megatronics" from reprapworld.com has a thermocouple input. This is through an MAX31855 thermocouple chip. Apparently the chip communicates using SPI. 

Am I correct in that Klipper doesn't support this yet?

KevinOConnor, 2018-02-16:
Klipper does not currently support the MAX31855 chip.  @cruwaller has a Klipper branch with thermocouple support, but I don't know what the state of it is.

closed 2018-02-25 by KevinOConnor

rewolff, 2018-02-25:
The referred "repository" is "behind" mainstream klipper quite a bit. We decided to downgrade the hardware instead. 

The merge was not trivial enough for me as a python-noob.

cruwaller, 2018-02-27:
Thermocouples are working just great . I am just using MAX31865 PT100 modules without problems.

My fork is behind and I have just taken a important fixes (if needed). I don't use/need LCD or simulated SD so skipped all those changes.... 
You need to do a merge if those are mandatory.