Klipper3d/klipper #172: Noob Question - no issues

Blue-Dust, 2018-02-14:
I am somewhat new to github, and a 3d printing noob.  (but have over 800 hours on my printer since receiving it late November).
I am currently planning to build a Hypercube/Core XY, and found a kit I am thinking to buy.
I want to make sure it will work Klipper before I buy it.
https://www.zyltech.com/store/p312/ZYLtech_HyperCube_Full_DIY_Kit_-_300mm_Cubic_Build_Volume_w%2F_Dual_Z.html
I will be upgrading the drivers to TMC2130s (looks like at the very least partially compatible now, maybe fully), and possibly even the steppers, but do not know to much about them or what out there is better just yet.  I am guessing everything else should work just fine.
I just want to make sure everything I buy will be compatible with Klipper before I buy, and not sure where I should be posting a question like this.  Please let me know if there is a better forum for non issue discussion, and I will post future questions/comments like this on there.
Also, if you have any suggestions on what parts I should be using, (and how to) please send it my way.

Thank you for your help!

jakep82, 2018-02-14:
I wouldn't recommend buying that kit unless you want to replace several components right away.  You want a 24V power supply for a bed that large, RAMPS won't support 24V without modifications, and DRV8825 steppers are really loud in my experience.  I've run an MKS Gen 1.4 with TMC2100 steppers on my Hypercube and it works great with a RPi3 and Klipper.  RUMBA is also a good option, and probably my top choice today because it has several 12V PWM outputs even with 24V input.

Blue-Dust, 2018-02-14:
I don't mind replacing components up to a point....
Is there a list of compatible/suggested parts I can choose from?  I have been trying to keep up on this project for a few months and very much new to 3D Printing components.

theopensourcerer, 2018-02-14:
For the control I also would recommend something like the MKS Gen 1.4. I'm in the process of building a Hypercube Evolution [1] at the moment and will be using this board with Klipper; with a 24v PSU.

For forums I suggest taking a look on Facebook. There are loads of groups for various printer styles & models. G+ has some 3D printing communities too.

[1] https://www.thingiverse.com/thing:2254103

Blue-Dust, 2018-02-14:
And really want to run Klipper.
Upfront, all I want to do is print, but plan to upgrade the printer over time add a few of the MK3 features and print with 4 filament colors (one hot end).
I wouldn't mind starting off with parts now to support that long term goal, but also won't feel bad replacing parts to get there either as Klipper becomes more mature.  I am just really impressed with this project and surprised others I have told about it who have had printers for years have not known of its existence. 
Thanks again for all suggestions!

jakep82, 2018-02-14:
Yeah, I would suggest Facebook for future questions.  I'm in the Hypercube group, and many knowledgeable people can answer questions there.  This isn't really the place for this conversation.

Blue-Dust, 2018-02-14:
I agree with you.  Just to confirm, is this the group you and others actively using Klipper joined? 
Thanks!
https://www.facebook.com/groups/1871049783171141?ref=m_notif&notif_t=group_comment&soft=bookmarks

jakep82, 2018-02-14:
Yes, and you can find several posts in that group about Klipper.

Blue-Dust, 2018-02-14:
Good stuff! That's the group I joined last week.  I have already posted a few Klipper/general 3D Printer questions on there.
Thanks!

closed 2018-02-14 by Blue-Dust