Klipper3d/klipper #174: Internal error on G28

sapell, 2018-02-14:
I updated Klipper to latest version available yesterday in order to get virtual sd card working.
Previous commit I had did not have that yet.
I now get internal error on G28 (timer too close) every time I try to home the machine, previous commit I had worked perfectly.
Apparently this error refers to the MCU not having enough resources with the newer commit, and the homing fails after a small nudge. I have 16x microstepping configured on DRV8825 drivers and BeagleBone Black running OctoPrint and Klipper, an Arduino Mega 2560 with RAMPS 1.4.
Is the newer commit that much heavier on the MCU than the one before virtual sd feature that now it can no longer even home? Log attached.
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1725839/klippy.log.gz)

KevinOConnor, 2018-02-15:
On Wed, Feb 14, 2018 at 10:33:09PM +0000, Simo Apell wrote:
> I updated Klipper to latest version available yesterday in order to get virtual sd card working.
> Previous commit I had did not have that yet.
> I now get internal error on G28 (timer too close) every [...]

This failure isn't related to virtual sd card.  I think the code is
struggling with the fact that your stepper_a has a slightly different
step distance than stepper_b and c.  I'm guessing that's an error in
your config.  I'll make a note of this, but could you retry with the
step_distance all set the same?

-Kevin

sapell, 2018-02-15:
That's would have been odd if it was. The step distances differ intentionally. I have measured each tower travel with 100mm movement and a caliper and adjusted the step distances accordingly. It's fairly tedious to adjust belt tensions such that each tower would move the same amount with same step distances. With previous commit I have successfully printed several prints with the exact same step distances. Unrelated and probably worth a separate issue, but could the config be structured such that each stepper home sensor position could be defined by offsets e.g. common z-height+offset? This would be beneficial for delta geometry and would make it easier to adjust z-height. Also z baby-stepping would be a great feature.

sapell, 2018-02-15:
Interesting, now it seems to home just fine and moves without error, I did not alter the step distances.

KevinOConnor, 2018-02-16:
On Thu, Feb 15, 2018 at 11:21:59AM -0800, Simo Apell wrote:
> It did it once again, error on G28 with 'timer too close' just after OctoPrint restart, but recovered after FIRMWARE_RESTART. Any suggestions what might be the reason? Do you think the Arduino Mega 2560 has too much to handle with 400steps/rev steppers, 16x microstepping and delta kinematics? As said, did not have any issues during printing prior to the update to the newest(-er) commit.

I suspect it is a very close timing race - probably not related to any
recent code change (nothing has changed recently in that area of the
code).

The homing process is a bit intricate - the code needs to generate all
the steps and load them entirely into the micro-controller before the
homing begins.  To facilitate this, the code rounds the homing speed
to an even multiple of clock ticks per step.  However, when homing a
delta, it's not really possible to do that when each stepper has a
different step distance.  In particular, it's rounding the homing
speed to stepper_a's step distance, and it's not getting good step
compression on stepper_b and stepper_c.

I need to look at it closer, but in the mean time, you should be able
to work around it by using the same step distance on all three
towers.  If you don't want to do that, you could instead try setting
HOMING_STEP_DELAY to 0.000002 at the top of homing.py.

-Kevin

sapell, 2018-02-23:
I have 'stubbornly' kept the differing step distances and changed HOMING_STEP_DELAY as instructed and have not encountered this issue after the change. Closing the issue as it seems the change of HOMING_STEP_DELAY has resolved the issue.

closed 2018-02-23 by sapell