Klipper3d/klipper #175: Print halted due to "MCU shutdown: Move queue empty" while printing from virtual SD

sapell, 2018-02-15:
Print just suddenly stopped while printing from virtual SD, Klipper commit 8fc3487.
Probably related with issue #160, but for me it occurred with BeagleBone Black + Arduino Mega + RAMPS setup.

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1729629/klippy.log.gz)

lucashpandolfo, 2018-02-15:
I interpret that klippy is sending klipper info too fast. When a stepper event is sent to klipper, a chunk of memory is allocated (removed from the ```move queue```) and the move info is stored there. When the move is completed, the memory is returned to the ```move queue``` (more like a memory pool than a queue I think :P). ```Move queue empty``` basically means that there is no memory left in the pool to 'store moves'. 

On the other hand, #160 seems to be another case of 'timer too close'. I don't think it's related.

sapell, 2018-02-15:
Thanks @lucashpandolfo  for the comment. If that really is the case, isn't there any mechanism for klippy to keep track of the move queue status to prevent this? Or might there be a memory leak there, if for some reason there are some 'crumbs' left of already completed moves?

Changed the title as the same occurred now when **not** using the virtual SD, log attached.

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1729753/klippy.log.gz)

KevinOConnor, 2018-02-16:
Very strange.  This is due to poor clock synchronization between host and mcu.  Similar to issue #160.  This one is much more dramatic though - the report:

Resetting prediction variance 16317.315: freq=15995459 diff=-545044 stddev=1365.696

indicates a clock drift of 545044 ticks in one second - 34 milliseconds - that's just crazy.

Anything in the system log at the time of the event?  (It may have been a system clock jump instead of an mcu clock jump.)

Besides this and #160 I haven't seen any reports like this.  I'm wondering if something strange is occurring on the Beaglebone.  What OS / kernel are you using?

Output of "scripts/graphstats -f" on your log:

[image]

sapell, 2018-02-16:
Hmm, what might be the cause of that large variation?

Here's the os and kernel info hostnamectl tells:
  Operating System: Debian GNU/Linux 9 (stretch)
            Kernel: Linux 4.4.91-ti-r133
      Architecture: arm

sapell, 2018-02-16:
Just noticed that this issue might have occurred because I changed feedrate to 150% during print. I tried the twice failed print again with older commit and noticed there was a ~2s pause of  in the movement in two consecutve layers when I had feedrate at 150%. When set feedrate back to 100% the pauses stopped occurring.

KevinOConnor, 2018-02-16:
On Fri, Feb 16, 2018 at 11:03:39AM +0000, Simo Apell wrote:
> Hmm, what might be the cause of that large variation?

I'm not sure - it's quite odd.  It could be an instability in the MCU
crystal, but that seems unlikely.  It could be a bug in Linux causing
bad timestamps to be returned (also unlikely - but a look at "speed
governor settings" could be warranted).  It could be NTP trying to
adjust the Linux clock (the monotonic clock isn't direclty impacted by
time changes, but if NTP is trying to speed/slow the clock to
synchronize, then maybe it could impact the monotonic clock).  It
could be a software defect in Klipper, but the time measurement code
paths are quite simple, so that doesn't seem likely.

On Fri, Feb 16, 2018 at 05:00:15PM +0000, Simo Apell wrote:
> Just noticed that this issue might have occurred because I changed feedrate to 150% during print. I tried the twice failed print again with older commit and noticed there was a ~2s pause of  in the [...]

I don't see how feedrate could impact the "move queue empty" issue -
feedrate is just a transformation of the incoming g-code coordinates.
Timing measurements are done in a separate thread in the host C code
(serialqueue.c).

A feedrate change will (of course) change the area of the extrusion
cross-section, and that will cause lookahead to flush its queue.  So,
a brief deaccel followed by accel wouldn't be unexpected.  Any other
impact to a feedrate change would be unexpected.

What was the previous commit id that did not show any issues for you?

-Kevin

sapell, 2018-02-16:
NTP could surely do that, should probably look in to disabling NTP or uninstall it altogether.

The commit that I now have is 21df21b. In issue #159 discussion I mentioned that I upgraded from Jan 12th 2018 commit, but that must have been my mistake as I could not have had my print cooling fans working properly with that commit (cycle_time was introduced in 21df21b).

Actually by feedrate I referred to the movement feedrate (M220, speed factor override percentage) as referred to in OctoPrint (Feed rate button). Octoprint refers with Flow rate to M221 extrude factor override percentage.

Now with 100% speed, it seems the print might even finish. Naturally OctoPrint is now munching the gcode as 21df21b does not yet support virtual SD.

KevinOConnor, 2018-02-16:
On Fri, Feb 16, 2018 at 06:29:21PM +0000, Simo Apell wrote:
> NTP could surely do that, should probably look in to disabling NTP or uninstall it altogether.

Another option would be to try changing
klippy/pyhelper.c:get_monotonic() to use CLOCK_MONOTONIC_RAW instead
of CLOCK_MONOTONIC.  I didn't do this originally, because it's unclear
to me all the implications of using this linux specific clock type.
However, if we get a bunch of testers showing there's no harm in using
it, then we could change to it.  (See "man clock_gettime" for the
distinction between the two clocks.)

-Kevin

sapell, 2018-02-16:
Thanks Kevin, once my print is complete, I could look in to this and checkout the latest commit also.

sapell, 2018-02-17:
One question, if I'm testing CLOCK_MONOTONIC_RAW, do I need to manually compile the pyhelper.c or does klippy compile it automatically when run? If manual compile is needed, is there a preferred way to do it, something klippy specific perhaps?

sapell, 2018-02-23:
It seems that either changing to CLOCK_MONOTONIC_RAW or that I have not used the speed multiplier adjustment during print has fixed the issue. At least I have not encountered this anymore. Closing the issue even though the root cause remains unresolved.

closed 2018-02-23 by sapell

KevinOConnor, 2018-02-23:
On Fri, Feb 23, 2018 at 01:48:45AM -0800, Simo Apell wrote:
> It seems that either changing to CLOCK_MONOTONIC_RAW or that I have not used the speed multiplier adjustment during print has fixed the issue. At least I have not encountered this anymore. Closing the issue even though the root cause remains unresolved.

Thanks.  CLOCK_MONOTONIC_RAW has been stable for you?

-Kevin

sapell, 2018-02-23:
Seems stable, I haven't encountered any issues whatsoever after switching to it.

sapell, 2018-04-22:
Just to let you know, this issue re-emerged after updating to commit bdd26e. Luckily it was during the first layer that it happened. I then noticed that I had forgotten to apply my stash which included the switch to CLOCK_MONOTONIC_RAW. Now the print has at least passed well beyond that halt point after I applied my stash.

KevinOConnor, 2018-04-23:
On Sun, Apr 22, 2018 at 01:05:53PM -0700, Simo Apell wrote:
> Just to let you know, this issue re-emerged after updating to commit bdd26e. Luckily it was during the first layer that it happened. I then noticed that I had forgotten to apply my stash which [...]

Okay, thanks.  Was the beaglebone recently booted up as well?  I
suspect this is related to ntp synchronization.  I did look a bit
further into CLOCK_MONOTONIC_RAW, and it appears that it's not as
"mainstream" as CLOCK_MONOTONIC, so I'm hesitant to convert over to it
for all users.

-Kevin

sapell, 2018-04-23:
Beaglebone wasn't booted recently, only updated Klipper as per the instructions in FAQ.