Klipper3d/klipper #176: Auto Bed Leveling

bjoern79de, 2018-02-16:
I've tried to configure auto bed leveling using the tilt section and have two issues:

1. I don't understand the relation of bed_tilt.probe_z_offset and stepper_z.position_endstop.
   When I set both to the same value, the hotend is way to high, but not by the sum of both.
   When I set one of them to 0, the Z-homing-behavior is also not the same as using just G28.

2. I get reproducable results from BED_TILT_CALIBRATE, but they are always too less to compensate the actual tilt of the bed. If I just manually probe two points along the x-axis and calculate the tilt, I get higher values than calculated by bed_tilt.py.

I'm using a inductive Z-Probe near the hotend (which is also the Z-endstop) and would like to automaticly set the tilt-adjustments automaticly before each print to compensate slight tilts between  two Z-steppers.

KevinOConnor, 2018-02-16:
There's little we can do without the klipper log file.  Please grab the latest code (which has some additional logging):

cd ~/klipper ; git pull ; sudo service klipper restart

and then run bed_tilt_calibrate, run m112, and then submit the log file as described at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

bjoern79de, 2018-02-21:
I guess, I've found out the reason for the tilt issue itself: My Z-Probe is out of the bed at X0,Y0, so I fetch Z0 at the middle of the bed within homing_override.

But the behavior of the Z-offsets are still weird.
If probe_z_offset / position_endstop are too high, and I step with G1 until I can move a paper under the nozzle and reduce the values about the current Z-position, the nozzle is way to high.

[klippy.log](https://github.com/KevinOConnor/klipper/files/1745283/klippy.log)

KevinOConnor, 2018-02-22:
On Wed, Feb 21, 2018 at 10:42:38AM -0800, bjoern79de wrote:
> I guess, I've found out the reason for the tilt issue itself: My Z-Probe is out of the bed at X0,Y0, so I fetch Z0 at the middle of the bed within homing_override.
>
> But the behavior of the Z-offsets [...]

Bed tilt calibration needs at least three points and you need to vary
both the x coordinate and y coordinate in the points probed.  Best is
to use a bunch of points.  See the config/sample-bltouch.cfg for some
ideas.

Your probe_z_offset should match your stepper_z position_endstop, and
it should be the distance between the nozzle and the bed when the
probe triggers.  If you perform the "paper test" and the nozzle is not
at z=0, then adjust both your probe_z_offset and position_endstop
accordingly.  So, for example, if an M114 reports the nozzle is at
z=1.0 when the paper test shows the nozzle close to the bed, then
increase both probe_z_offset and position_endstop by 1mm.

-Kevin

bjoern79de, 2018-02-22:
I just use 2 points to simplify debugging and only focus on the x-axis.
It works after moving X0 and measure Z0 there - the nozzle follows the
X-tilt correctly. The nozzle does not touch the bed during probing. My
Problem is, that I don't understand why I need two z-offsets for one
probe/endstop and why changing the offsets by 1mm does Not Change the
nozzle height by 1mm.

When M114 says the nozzle ist at 1mm and it touches the bed and I decrease
both offsets by 1mm, the nozzle is around 0.6mm above the bed the next
time. It can't be related to calibration or sensor variance, because the
distance ist always the same and when I set the  offsets by trial&error, a
printed object have the expected height.

> I guess, I've found out the reason for the tilt issue itself: My Z-Probe
is out of the bed at X0,Y0, so I fetch Z0 at the middle of the bed within
homing_override.
>
> But the behavior of the Z-offsets are still weird.
> If probe_z_offset / position_endstop are too high, and I step with G1
until I can move a paper under the nozzle and reduce the values about the
current Z-position, the nozzle is way to high.

Bed tilt calibration needs at least three points and you need to vary
both the x coordinate and y coordinate in the points probed. Best is
to use a bunch of points. See the config/sample-bltouch.cfg for some
ideas.

Your probe_z_offset should match your stepper_z position_endstop, and
it should be the distance between the nozzle and the bed when the
probe triggers. If you perform the "paper test" and the nozzle is not
at z=0, then adjust both your probe_z_offset and position_endstop
accordingly. So, for example, if an M114 reports the nozzle is at
z=1.0 when the paper test shows the nozzle close to the bed, then
increase both probe_z_offset and position_endstop by 1mm.

-Kevin

KevinOConnor, 2018-02-22:
On Thu, Feb 22, 2018 at 08:14:16AM +0000, bjoern79de wrote:
> I just use 2 points to simplify debugging and only focus on the x-axis.
> It works after moving X0 and measure Z0 there - the nozzle follows the
> X-tilt correctly. The nozzle does not touch the bed [...]

The code is new, so it's possible there is a bug in it.  However, I
don't have enough information to track down what it is you are seeing.

The code isn't going to work right with only two points (it isn't
simplifying debugging).  Put in at least three points (preferably use
a bunch) and make sure to vary both x and y.

Then set both position_endstop and probe_z_offset to the same value
(eg, 2.345) and run BED_TILT_CALIBRATE.  Update the config with the
suggested values, run RESTART, and if that results in a value too far
from the bed, adjust both position_endstop and probe_z_offset
accordingly, issue a RESTART command, and then rerun
BED_TILT_CALIBRATE.  Then attach the klippy.log file (which should
contain the three sessions and both BED_TILT_CALIBRATE commands).  I
can then look at that log file to see what went wrong.

-Kevin

KevinOConnor, 2018-03-18:
I believe the main underlying issue here has been fixed.  Issue #227 has further details.

closed 2018-03-18 by KevinOConnor