Klipper3d/klipper #177: What am I doing wrong.....MCU error

itri4ever, 2018-02-16:
I have tried almost everything to resolve this issue.  No idea where to start troubleshooting this issue....any help is greatly appreciated.
[Klippy Log.txt](https://github.com/KevinOConnor/klipper/files/1732426/Klippy.Log.txt)

KevinOConnor, 2018-02-16:
On Fri, Feb 16, 2018 at 10:36:46AM -0800, itri4ever wrote:
> I have tried almost everything to resolve this issue.  No idea where to start troubleshooting this issue....any help is greatly appreciated.

Unfortunately, your log file was corrupted.  Be sure to use an scp
program when transferring between the Raspberry Pi and your desktop.

Best guess is that you don't have a thermistor attached to the
analog14 line.  Try commenting out the entire heater_bed section - if
that starts up, then you'll need to track down the correct bed heater
wiring for your printer.

-Kevin

itri4ever, 2018-02-16:
Ill try that - but in the meantime, here is the actual error:

Recv: // MCU 'mcu' shutdown: ADC out of range
Recv: // This generally occurs when a heater temperature exceeds
Recv: // its configured min_temp or max_temp.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: !! Printer is shutdown

itri4ever, 2018-02-16:
well commenting out the bed heater worked - started up

im assuming by when you say track down the correct bed heater wiring, you mean change the thermostat name to the correct on in the printer.cfg file?

KevinOConnor, 2018-02-22:
No - it's likely the thermistor pin is incorrect. (Or the thermistor is not functioning correctly.)  You'll need to track down the correct wiring from the printer manufacturer.

closed 2018-02-25 by KevinOConnor