Klipper3d/klipper #178: Feature request: Expressing position_endstop in printer.cfg as a formula

sapell, 2018-02-16:
As a delta printer user, I'd very much like to have the option to express the position_endstop as a formula in e.g. `position_endstop: common_z_height + endstop_offset`, where the common_z_height would be common for all towers and endstop_offset per each tower. This way adjusting z-height would be at least slightly more convenient. The z-height adjustment is not needed that often, but for instance when printing with PETG, it's often necessary to have the nozzle a bit higher than for PLA for instance.

lenne0815, 2018-02-24:
Couldnt you do the exact same thing even more convenient by setting different initial layer heights in you slicer material profile ? ( Maybe im missing something here but isnt that exactly the same ? )

sapell, 2018-02-24:
That would also change the amount of extruded plastic. For PET-G it's necessary to let it lay down from slightly higher than e.g. PLA, otherwise the nozzle picks up a glob of PET-G quite quickly as it prints. Try PET-G, you'll see what I'm talking about. ;)

lenne0815, 2018-02-24:
Which you can easily counteract by lowering the extrusion multiplier for the first layer ( For S3D this feature is called First layer width, for cura Initial layer line width )

sapell, 2018-02-24:
That would not be a solution to the problem IMO, not buing that. I use Slic3r, had bad experiences with Cura, S3D is too expensive. Coming from Repetier, babystepping is a feature I miss from it (see my other feature request). If babystepping was included in Klipper, this feature would not be needed.

lenne0815, 2018-02-24:
With a likelyhood of 115% sli3er has exactly the same feature. This specific feature is not needed at all. I do agree with the live Z adjustment part tough, that would be neat.

sapell, 2018-02-24:
Yeah, Slic3r has the First layer extrusion width setting.

cmock, 2018-02-25:
What does remain, even when you're able to set up your material-dependent offsets by some workaround, is the inconvenience of adjusting z-height on a delta -- you have to add/substract the same value from each tower's ``position_endstop``. 

That is error-prone and makes the user do work that a computer can do much better.

(I have to adjust my z-height regularily, be it due to slight variances when changing build plates, or because my printer's mechanics are not as stable as they're supposed to be).

KevinOConnor, 2018-02-26:
The M206 command can be used to apply a Z offset.  That said, M206 can be a bit cryptic to use.

I'm open to changing the config - it would help if I got a pull request with the config and documentation changes for it.

sapell, 2018-02-26:
I'll try and see if the M206 does the trick, maybe it's not worth it to add more complexity to the config file as it would serve only delta robot users.

closed 2018-02-26 by sapell