Klipper3d/klipper #181: CR10 cfg file change suggestion

dionm9, 2018-02-17:
Hello,
I wasn't sure where to post this, but after a few hours of problem solving, I found out that the printer-creality-cr10-2017.cfg file needs the line
baud:115200
I was just wondering if that could be added to make made installation easier for other CR10 users.

KevinOConnor, 2018-02-17:
On Sat, Feb 17, 2018 at 09:50:24PM +0000, dionm9 wrote:
> Hello,
> I wasn't sure where to post this, but after a few hours of problem solving, I found out that the printer-creality-cr10-2017.cfg file needs the line
> baud:115200
> I was just wondering if that could be added to make made installation easier for other CR10 users. 

The recommended baud rate is 250000.  You should rerun "make
menuconfig", set the baud back to its default of 250000, reflash the
micro-controller, and then use the default of 250000 in your
printer.cfg file.

If after doing the above, you encouter a problem, you should post the
klipper log file as described at:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

nerobot, 2018-02-18:
I can confirm that a baudrate of 250000 works fine in my CR10 without any obvious issues.

Steven

closed 2018-02-25 by KevinOConnor