Klipper3d/klipper #182: Missed scheduling of next pca9685 event

jakep82, 2018-02-18:
This is the first time I've seen this issue.  Replicape with BBB wireless running Jessie console.  I do see a few errors when checking the klipper_pru service.  The ADC error doesn't surprise me as the slots file has been deprecated.  The other 2 errors are unfamiliar.  Not sure if this could be causing an issue.  I've had several successful prints since switching to console.  This is the first failure.

> Feb 17 22:47:54 beaglebone systemd[1]: Starting LSB: Klipper_PRU daemon...
> Feb 17 22:47:55 beaglebone klipper_pru[795]: sh: echo: I/O error
> Feb 17 22:47:55 beaglebone klipper_pru[795]: sh: echo: I/O error
> Feb 17 22:47:56 beaglebone klipper_pru[795]: Starting pru:Loading ADC module:/etc/init.d/klipper_pru: 57: /etc/init.d/klipper_pru: cannot create /sys/devices/platform/bone_capemgr/slots: Permission denied
> Feb 17 22:47:56 beaglebone klipper_pru[795]: Starting klipper MCU: klipper_pru.
> Feb 17 22:47:56 beaglebone systemd[1]: Started LSB: Klipper_PRU daemon.

> Recv: // MCU 'host' shutdown: Missed scheduling of next pca9685 event
> Recv: // This is generally indicative of an intermittent
> Recv: // communication failure between micro-controller and host.

[klippy.log](https://github.com/KevinOConnor/klipper/files/1734994/klippy.log)

KevinOConnor, 2018-02-20:
On Sun, Feb 18, 2018 at 07:06:27PM +0000, jakep82 wrote:
> This is the first time I've seen this issue.  Replicape with BBB wireless running Jessie console.  I do see a few errors when checking the klipper_pru service.  The ADC error doesn't surprise me as [...]

This appears to be the result of a subtle problem in the clocksync
code.  I think it should be fixed now (commit 02ae2ab9).

-Kevin

jakep82, 2018-03-03:
This doesn't appear to be solved.  I did a git pull and restarted Klipper tonight.  The same gcode produced the same error.  Log attached.
[klippy.log](https://github.com/KevinOConnor/klipper/files/1777193/klippy.log)

KevinOConnor, 2018-03-04:
On Sat, Mar 03, 2018 at 06:51:27AM +0000, jakep82 wrote:
> This doesn't appear to be solved.  I did a git pull and restarted Klipper tonight.  The same gcode produced the same error.  Log attached.

There was an error in commit 02ae2ab9.  It should hopefully be fixed
now (commit 7290ed5f).

-Kevin

jakep82, 2018-03-06:
Latest commit appears to have solved this issue.

closed 2018-03-06 by jakep82