Klipper3d/klipper #183: Replicape klipper problems getting started

marcuswu, 2018-02-20:
I am having numerous issues, but first, I have started from scratch twice following the directions from the [install doc](https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md), [beaglebone doc](https://github.com/KevinOConnor/klipper/blob/master/docs/beaglebone.md), and then the [config check doc](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md). I am running the latest (checked out from here) code on a Replicape B3A connected to a Beaglebone Black. First, I had tried bone-debian-9.3-iot-armhf-2018-01-28-4gb.img.xz, but I am currently running bone-debian-8.7-iot-armhf-2017-03-19-4gb.img.xz because the install docs say to use the latest Jesse IoT image.

Things I've found during config check:
* The temperatures in Octoprint report fine
* The one motor moves freely by hand and the rest are under power (don't move)
* The X endstop reports that it is triggered regardless of whether or not I am pressing it
* Attempting to home any axis results in an error (MCU shutdown). Log attached starting from a reset. If you need a full log from boot up to error, let me know.
* I wasn't sure how to calculate step_distance for steppers configured for spread16. Spread16 does step interpolation and I wasn't sure whether to calculate for 16 microsteps or for the interpolated microsteps. I guessed the calculation should be for 16 microsteps.

I've attached my config and my klippy log for the homing error.

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/1740396/printer.cfg.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/1740327/klippy.log)

KevinOConnor, 2018-02-20:
On Tue, Feb 20, 2018 at 02:27:55PM +0000, Marcus Wu wrote:
> I am having numerous issues, but first, I have started from scratch twice following the directions from the [install doc](https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md), [...]

Which motors move freely and which are powered?  Can you confirm that
they are powered even after a full restart and prior to issuing any
commands in OctoPrint?

> * The X endstop reports that it is triggered regardless of whether or not I am pressing it

Do the other endstops work?  Are you sure the connections are good and
on the right ports?  Otherwise, the only other thing I can think of is
a "device tree overlay" problem - something not setting the pull up or
not setting the port as input.

> * Attempting to home any axis results in an error (MCU shutdown). Log attached starting from a reset. If you need a full log from boot up to error, let me know.
> * I wasn't sure how to calculate [...]

Yes - divide the full step distance by 16 when using spread16.

> I've attached my config and my klippy log for the homing error.

This appears to be a problem with "homing speed rounding" on corexy.
I need to think about how to fix this.  In the mean time, you should
remove the "homing_speed = 300" from your config, and double check
your x/y step distance (.000625 is very small for a corexy axis).  I
suspect it will work with a slower homing speed and/or with a larger
step distance.

-Kevin

marcuswu, 2018-02-20:
> Which motors move freely and which are powered?  Can you confirm that they are powered even after a full restart and prior to issuing any commands in OctoPrint?

All but the Y-axis motor are powered

> Do the other endstops work?  Are you sure the connections are good and on the right ports?
 Otherwise, the only other thing I can think of is a "device tree overlay" problem - something not setting the pull up or not setting the port as input.

Yes, the other endstops work. I am sure the connections are good and on the correct ports. I can also confirm that the light on the x-axis endstop does turn on / off when pressing and releasing it. How can I check the device tree overlay?

> This appears to be a problem with "homing speed rounding" on corexy. I need to think about how to fix this.  In the mean time, you should remove the "homing_speed = 300" from your config, and double [...]

Ok, I'm going to redo my calculations -- very possible I botched that.

jakep82, 2018-02-20:
On my machine with 20mm timing pulleys and 1.8 degree steppers, my step distance in spread16 mode is .00625.  I think you've probably just got an extra zero.  Also, I don't think the TMC2100 steppers on the Replicape are powerful enough to move any printer anywhere close to your max 1500mm/s, and you might lose steps at even the 300mm/s you have called out for homing.

ifUpEth0, 2018-02-21:
I am on a Cartesian double z.
I am on the same version of replicape and debian IoT. I am experiencing the same issue with the x endstop and M84 not working. I have verified the pins for x. I can try to move to x2, but this setup works when booted under redeem right now. It is possible that tone of my z axis moves free, but cannot really check the z. X and Y will not free up. I have not tried
[klippy.log](https://github.com/KevinOConnor/klipper/files/1742412/klippy.log)
 to home my machine yet.

KevinOConnor, 2018-02-21:
Do fans and heaters work?  What does the following report?  ls -l /dev/spi*

ifUpEth0, 2018-02-21:
Fans and heaters work. I should have been more clear y and z endstops work as well. X is the only one that does not work.

debian@beaglebone:~$ ls -l /dev/spi*
crw-rw---- 1 root spi 153, 1 Feb 21 16:35 /dev/spidev1.0
crw-rw---- 1 root spi 153, 0 Feb 21 16:35 /dev/spidev1.1
crw-rw---- 1 root spi 153, 3 Feb 21 16:35 /dev/spidev2.0
crw-rw---- 1 root spi 153, 2 Feb 21 16:35 /dev/spidev2.1
debian@beaglebone:~$

ifUpEth0, 2018-02-21:
I am going to boot back into redeem and grab the /boot/uEnv.txt to compare. Are there any other files related to cape management I should grab?

KevinOConnor, 2018-02-21:
On Wed, Feb 21, 2018 at 04:38:54PM +0000, ifUpEth0 wrote:
> I am going to boot back into redeem and grab the /boot/uEnv.txt to compare. Are there any other files related to cape management I should grab?

What does:

cat /sys/devices/platform/bone_capemgr/slots
ls -l /dev/spi*

report on both installs?

-Kevin

marcuswu, 2018-02-21:
Fans and heaters work for me as well.

```
$ ls -l /dev/spi*
crw-rw---- 1 root spi 153, 1 Feb 21 00:26 /dev/spidev1.0
crw-rw---- 1 root spi 153, 0 Feb 21 00:26 /dev/spidev1.1
crw-rw---- 1 root spi 153, 3 Feb 21 00:26 /dev/spidev2.0
crw-rw---- 1 root spi 153, 2 Feb 21 00:26 /dev/spidev2.1
```

marcuswu, 2018-02-21:
I went over and checked out #185 and I want to add that I did have to enable SPI earlier in my setup process. That issue references these changes:
```
cape_disable=bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN
cape_enable=bone_capemgr.enable_partno=BB-SPIDEV0,BB-SPIDEV1
```

I only have the bottom line -- I didn't seem to have to disable HDMI explicitly to get /dev/spidev2.0 and /dev/spidev2.1

ifUpEth0, 2018-02-21:
My findings so far.

Welcome to Ubuntu 16.04.3 LTS (GNU/Linux 4.4.69-bone17 armv7l)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
Last login: Mon Feb 19 18:40:27 2018 from 192.168.86.187
root@kamikaze:~# uname -r
4.4.69-bone17

root@kamikaze:~# cat /sys/devices/platform/bone_capemgr/slots
 0: P---L-   0 Replicape 3D printer cape,0B3A,Intelligent Agen,BB-BONE-REPLICAP
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1

root@kamikaze:~# ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Feb 20 23:18 /dev/spidev2.1

Diff in uEnv.txt

Kamikazi:
+ uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
+ enable_uboot_cape_universal=1
	- cape_universal=enable
+ consoleblank=0 fbcon=rotate:1 omap_wdt.nowayout=0 root=/dev/mmcblk0p1 root=/dev/mmcblk0p1
[uEnv_klipper.txt](https://github.com/KevinOConnor/klipper/files/1745333/uEnv_klipper.txt)
[uEnv_redeem.txt](https://github.com/KevinOConnor/klipper/files/1745336/uEnv_redeem.txt)

KevinOConnor, 2018-02-22:
@jakep82 what does your beaglebone report for:
ls -l /dev/spi*
cat /sys/devices/platform/bone_capemgr/slots

marcuswu, 2018-02-22:
Forgive me if my thinking here is flawed -- I'm new to how this operates. I was thinking yesterday that it'd be interesting to try klipper with the dtbo that is installed with Umikaze. I have a feeling that dtbo is kernel specific, though so I tried an experiment yesterday to try running klipper from Umikaze. I managed to get everything set up and installed, but got stopped because Umikaze doesn't come with the pru-rproc driver.

KevinOConnor, 2018-02-22:
I'd hoped there would be one somewhat standardized distribution for the beaglebone that could be used.  I'm sure other distros could be made to work, but it's too time consuming for me to check them out.

Separately, if you modify klippy/extras/replicape.py to set REPLICAPE_SHIFT_REGISTER_BUS = 2 - does that help?  (You'll need to run "sudo service klipper restart" after making the change.

ifUpEth0, 2018-02-22:
Changing  REPLICAPE_SHIFT_REGISTER_BUS = 2 give me a working M84, but x endstop is still misbehaving. 

I then moved the x endstop to x2 pin p9_11. With this config x endstop is working however M84 and M18 are failing to release the motors now. Homing X and Y seemed to work. I am now trying to work out issues with my dual z. This is looking promising! Thanks a lot.

ifUpEth0, 2018-02-22:
It seems I can only get one fan pin at a time working. Last in wins? Idealy I could run them like under redeem and say m106 p2 s255, but any m106 codes regardless of p value only control one fan the last in this series below from the config file.

[fan]
pin: replicape:power_fan0
pin: replicape:power_fan2
pin: replicape:power_fan3

I may hard wire the coldend and board cooling fans for now. That would be the only way to move forward with a test print.

KevinOConnor, 2018-02-22:
On Thu, Feb 22, 2018 at 10:31:14PM +0000, ifUpEth0 wrote:
> Changing  REPLICAPE_SHIFT_REGISTER_BUS = 2 give me a working M84, but x endstop is still misbehaving.
>
> I then moved the x endstop to x2 pin p9_11. With this config x endstop is working however M84 [...]

Did you reboot between moving to the x2 pin?  It's possible the spi
device went back to bus 1.  What does 'ls -l /dev/spi*' show now?

Also, the last config you sent had your z1 using
replicape:stepper_z_enable, but that should have been
replicape:stepper_h_enable.

-Kevin

KevinOConnor, 2018-02-22:
On Thu, Feb 22, 2018 at 11:08:26PM +0000, ifUpEth0 wrote:
> It seems I can only get one fan pin at a time working. Last in wins? Idealy I could run them like under redeem and say m106 p2 s255, but any m106 codes regardless of p value only control one fan the [...]

If you want to control all three fans simultaneously, you can
configure a [multi_pin] config section.  If you want to enable a fan
when one of the heaters are on, then use a [heater_fan] section.
Finally, if you want to manually control them via the SET_PIN command,
then configuire [pwm_output] sections.

-Kevin

ifUpEth0, 2018-02-22:
Thanks Kevin. Your keywords lead me to this file https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg

About to run a test print with the following config

[printer.cfg (2).txt](https://github.com/KevinOConnor/klipper/files/1749966/printer.cfg.2.txt)

marcuswu, 2018-02-23:
I loaded up debian with klipper again, reconfigured everything, and then set REPLICAPE_SHIFT_REGISTER_BUS = 2, but that didn't seem to do anything for M84.

marcuswu, 2018-02-23:
On the plus side, I swapped the X endstop pin to X2 following @ifUpEth0's lead and that is now working. I can now home and move all axes!

jakep82, 2018-03-03:
Sorry for the late reply.  

debian@beaglebone:~$ ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Feb 18 11:07 /dev/spidev1.1

debian@beaglebone:~$ cat /sys/devices/platform/bone_capemgr/slots
cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory

As previously noted I'm on Jessie console, and my printer seems to work fine 99% of the time.

KevinOConnor, 2018-03-12:
I'd like to make sure I understand the final state of this issue:
- some users can't use the X endstop pin on their replicape, while other users with (seemingly identical hardware) can
- some users (the same users?) stepper motors are always on
- there is wide variety in how many spidev devices are created, and in at least some cases, the spidev device that we want isn't spidev1.1
Is this correct?

jakep82, 2018-03-12:
My system details are above, and for me everything seems to be working.  Xmin, Ymin, and Zmin endstops all work as expected.  I haven't tested the other endstops as I don't need them currently.  I'm also able to completely disable X, Y, Z, and E steppers with M18 or M84.  I doubt it's important, but my printer is a coreXY.

The only thing I haven't got working yet is my BLTouch.  The endstop portion is fine, but I haven't been able to get the servo portion working.  This could be a setup issue on my end and I haven't had time to troubleshoot, so I'm still using a regular endstop for Z.

KevinOConnor, 2018-03-12:
On Mon, Mar 12, 2018 at 07:01:26PM +0000, jakep82 wrote:
> My system details are above, and for me everything seems to be working.  Xmin, Ymin, and Zmin endstops all work as expected.  I haven't tested the other endstops as I don't need them currently.  I'm [...]

Thanks.

> The only thing I haven't got working yet is my BLTouch.  The endstop portion is fine, but I haven't been able to get the servo portion working.  This could be a setup issue on my end and I haven't had time to troubleshoot, so I'm still using a regular endstop for Z.

If you've connected the servo to X2 or Y2 then there's a separate
servo enable system on the replicape that I didn't code.  If so, open
a separate issue and I'll put together some test code to enable it.

-Kevin

jakep82, 2018-03-12:
I opened an issue.  I tried both X2 and Y2, but I wasn't aware they needed servo support enabled with additional code.

closed 2018-03-26 by KevinOConnor