Klipper3d/klipper #185: MCU unrecoverable replicape

ifUpEth0, 2018-02-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1741424/klippy.log)

image: bone-debian-8.7-iot-armhf-2017-03-19-4gb
replicape: B3A

From a fresh install I can get octoprint and klipper talking and can see what look to be accurate temps. Every time I get to the step "M112" the system will not recover. I have of course entered "FIRMWARE_RESTART" but am met with T:0 and a status reply of

Recv: // MCU 'host' error during config: Unable to open spi device
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: !! Error configuring printer

I have had this same hardware running on klipper with ramps prior. I have the onboard flashed with redeem and can boot into klipper at any time to test. Would love to cut over to klipper as I am really struggling to get good prints out of redeem.

jakep82, 2018-02-20:
Try a newer snapshot build of Jessie iot from the link below.  Someone else was having an issue with /dev/rpmsg_pru30, but I've not seen this issue on recent snapshots of Jessie iot or console.

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_iot

KevinOConnor, 2018-02-20:
On Tue, Feb 20, 2018 at 06:56:29PM +0000, ifUpEth0 wrote:
> image: bone-debian-8.7-iot-armhf-2017-03-19-4gb
> replicape: B3A
>
> From a fresh install I can get octoprint and klipper talking and can see what look to be accurate temps. Every time I get to the step [...]

That's odd.  What does 'ls -l /dev/spidev*' report?  There should be
an spi device created by the OS.

-Kevin

ifUpEth0, 2018-02-20:
ls -l /dev/spidev* gives error. It would seem the spi device is missing. I see nothing like it in /dev.
I can try a new snapshot but when I tried bone-debian-9.3-iot-armhf-2018-01-28-4gb I could never get past the "/dev/rpmsg_pru30" error mentioned above.

ifUpEth0, 2018-02-20:
the spi tip helped. Should anyone else end up here. Check /boot/uEnv.txt.
You probably need the following:
cape_disable=bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN
cape_enable=bone_capemgr.enable_partno=BB-SPIDEV0,BB-SPIDEV1
After this and a reboot I again have my temps and can M112 and FIRMWARE_RESTART as expect.

KevinOConnor, 2018-02-21:
On Tue, Feb 20, 2018 at 11:51:38PM +0000, ifUpEth0 wrote:
> the spi tip helped. Should anyone else end up here. Check /boot/uEnv.txt.
> You probably need the following:
> cape_disable=bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN [...]

The "cape manager" stuff on the beaglebone is a complete nightmare.
:-(

I suspect issue #183 is a similar (but different) mysterious problem
with the cape manager.

-Kevin

closed 2018-02-25 by KevinOConnor