Klipper3d/klipper #188: Multiple extruders using single fan (Chimera print head)

pyr0ball, 2018-02-22:
I have a hotend that uses the same fan for both extruders, and would like to be able to control it using the [heater fan] function, however I wasn't sure if I should make a second section for each extruder, or put them both under the same section somehow. Anyhow making a separate section didn't work based on the logs, as it doesn't like having the same pin assigned twice it seems.

I changed it to this:

```
[heater_fan extruder fan]
pin: ar2
heater: extruder extruder1
heater_temp: 40.0
fan_speed: 1.0
```
 which doesn't appear to be crashing anything, but I won't know til I get the board hooked up if it works as intended. If it does, I think adding a comment about this in the examples would be a good idea

[klippy.log](https://github.com/KevinOConnor/klipper/files/1747040/klippy.log)

KevinOConnor, 2018-02-22:
On Thu, Feb 22, 2018 at 12:05:20AM -0800, pyr0ball wrote:
> I have a hotend that uses the same fan for both extruders, and would like to be able to control it using the [heater fan] function, however I wasn't sure if I should make a second section for each [...]

There is no current way to enable a fan when any one of multiple
heaters are on.  You could tie it to just one of the heaters, you
could update the code to support it, or you could use a pwm_output
config section and leave the pin on all the time (or manually control
it with the SET_PIN command).

> I changed it to this:
> 
> ```
> [heater_fan extruder fan]
> pin: ar2
> heater: extruder extruder1
> heater_temp: 40.0
> fan_speed: 1.0
> ```

That just produces an error for me (as it should).

-Kevin

pyr0ball, 2018-02-26:
Once I get better at python I should be able to contribute at a code level, but for now I think what I'll end up doing it wiring up the fan ground pin to two different pins and configuring each of those for each heater. Should have the same effect.

You can go ahead and close this for now

closed 2018-03-02 by KevinOConnor