Klipper3d/klipper #189: How to call function on idle?  Power down PSU.

wizhippo, 2018-02-23:
I'm looking in the source and cannot pinpoint it.  I would like to disable a pin I have connected to a psu when the printer is idle for x amount of time.

Can you point me to where I may add this?

lenne0815, 2018-02-24:
Probably doing it from the raspi i/o is the way to go there, ruling out communication issues for safety: this is what a quick google brought up https://github.com/foosel/OctoPrint/wiki/Controlling-a-relay-board-from-your-RPi

wizhippo, 2018-02-24:
Either way, what I am looking for is in Klipper where is idle being detected to turn the motors off?

lenne0815, 2018-02-24:
Just check where the ena pin from the stepsticks is called, it wont help much for switching a psu though as the motors are idle on boot. When you just use a psu for exclusively for the stepsticks i doubt you can switch it fast enough to not loose steps on the first movement when waking up.

lenne0815, 2018-02-24:
Or add a G4 + SET_PIN PIN=my_pin VALUE=0 ( [digital_output my_pin] ) to switch a pin after your gcode.

KevinOConnor, 2018-02-24:
On Fri, Feb 23, 2018 at 10:48:28AM -0800, Douglas Hammond wrote:
> I'm looking in the source and cannot pinpoint it.  I would like to disable a pin I have connected to a psu when the printer is idle for x amount of time.
> 
> Can you point me to where I may add this?

There isn't really a mechanism in the code for this today.  The
closest would be the motor_off_timer / motor_off_handler in
toolhead.py.

However, if this powers off the mcu, you'd want to notify the mcu code
so that it doesn't throw a "comms loss" error.  Then there would be
the work needed to bring it all back up on activity.

-Kevin

cmock, 2018-02-25:
One solution that works: using the "PSU control" plugin in Octoprint, and a pin configured in klipper for the PS_ON signal on the RAMPS:
```
[digital_output power]
pin: !ar12
value: 0
shutdown_value: 0
```
Then, the plugin can use ``SET_PIN PIN=power VALUE=1`` to turn on the PSU. Basically, I'm replicating the M80/M81 commands used in other firmwares to switch a PSU.

I have my PSU, electronics and hotend fans plus the LED lights wired up directly to the +12V rail, so the printer is really quiet until it actually does work. The plugin does the "idle" handling, including a threshold for the hotend temperature so the hotend fan is not turned off too soon.

The arduino/RAMPS/LCD is powered via USB from the Raspberry, so it is runnning all the time.

The plugin, BTW, could also use a Raspberry GPIO pin to switch the PSU.

wizhippo, 2018-02-25:
Psu control with its idle timeout looks like what I am looking for thank you.

closed 2018-02-25 by wizhippo