Klipper3d/klipper #192: Request for 'user guide'

sapell, 2018-02-23:
I really appreciate the existing documentation and understand that composing a complete user guide is an overwhelming task, but I'd still like to see some document that would draw together the basic usage and could also act as a quick reference. For instance it could include what parameters Klipper expects for commands, e.g. how to use PID_TUNE/M303. I took the M303 as an example as it has different implementations on different firmwares, it's difficult for the user to guess what 'standard' Klipper is following if any.

lenne0815, 2018-02-23:
I attempted to do something like that here http://www.lennesblog.de/2017/11/14/klipper-firmware-pt1-why/ it should atleast cover the basics to get you up and running, keep in mind that thats my personal blog and im extremely biased towards klipper and used every opportunity to take jabs at other firmwares ;)

mmone, 2018-02-23:
I put together an unofficial list of [gcodes](https://github.com/mmone/klipper/tree/command-docs/docs/gcode) and [commands](https://github.com/mmone/klipper/tree/command-docs/docs/commands) supported by Klipper.

sapell, 2018-02-24:
Thanks lenne0815 and mmone! I consider the request fulfilled, closing issue.

closed 2018-02-24 by sapell