Klipper3d/klipper #193: Support for multiple carriages

w1ebr, 2018-02-24:
Kevin, I have an IDEX machine where there are 2 x-axis carriages and a single y-axis for both. Syntactically, I see how a [stepper_x1] section could be created for the second carriage and [extruder1] for the second extruder. Would the stepper_x1 section have something like uses: extruder1 in it? (or vice versa, with the extruder section pointing back to the related stepper, or both) OR maybe a [tool] section where the set of x, y, z and e steppers for a given tool is defined?

I am asking this question before asking the more obvious question about whether it is supported because I have seen how quickly you address new areas of development :-)     ( I hope you are smiling!)

Thanks for everything you have created so far!

Gene

KevinOConnor, 2018-02-25:
On Sat, Feb 24, 2018 at 03:54:45AM +0000, w1ebr wrote:
> Kevin, I have an IDEX machine where there are 2 x-axis carriages and a single y-axis for both. Syntactically, I see how a [stepper_x1] section could be created for the second carriage and [...]

There isn't currently support for "dual x extruders".  I don't think
it would be too difficult to add.

I can probably put together some experiemental code if you're up for
testing.  If so, get the printer fully working with one extruder and
then post the klippy.log file.  Leave one of the x carriage motors
disabled in that config, but add to this issue the stepper section
that you would use to describe it.

-Kevin

w1ebr, 2018-02-25:
OK, will do!

[stepper_x1]

# The stepper_x section is used to describe the stepper controlling
# the X axis in a cartesian robot.

step_pin: ar29
dir_pin: !ar28
enable_pin: !ar39
step_distance: .0100
endstop_pin: ^ar37
#position_min: -44
position_endstop: -44
position_max: 324
homing_speed: 50.0
homing_retract_dist: 5.0
homing_positive_dir: false

And this was my other idea:

[tool]
uses: stepper_x, stepper_y, stepper_z, extruder

[tool1]
uses: stepper_x1, stepper_y, stepper_z, extruder1

Gene

w1ebr, 2018-03-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1772663/klippy.log)

OK, that's everything except the bed leveling working (which I still don't understand why it doesn't converge yet, but 1 issue at a time )

Thanks!

KevinOConnor, 2018-03-02:
Okay - can you try the work-dual-carriage-20180302 branch?

cd ~/klipper ; git fetch ; git checkout origin/work-dual-carriage-20180302 ; sudo service klipper restart

For now, the config uses a [dual_carriage] section.  You would use the new SET_DUAL_CARRIAGE command to activate a carriage.  See the updated config/example-extras.cfg for the details.  I don't have a good way to test this, so there's a good chance it wont work.

w1ebr, 2018-03-02:
Wow! Alright!

KevinOConnor, 2018-03-03:
> The only thing left is to handle the different nozzle offsets ( the carriage nozzles are not at exactly the same height and both are offset from the probe endstop )

Use the M206 command in the extruder activate_gcode / deactivate_gcode sections to account for nozzle offsets.

-Kevin

w1ebr, 2018-03-09:
Hi Kevin, has this and the LCD code been moved to master?

KevinOConnor, 2018-03-09:
On Fri, Mar 09, 2018 at 01:43:03PM +0000, w1ebr wrote:
> Hi Kevin, has this and the LCD code been moved to master?

I just merged dual carriage support into master (commit 4d48c111).
The LCD code is also on master now.

-Kevin

w1ebr, 2018-03-09:
Thank you so much!

closed 2018-03-09 by w1ebr

SrGeek, 2019-08-24:
Is there a way to Query carriage states?  I am trying to troubleshoot my parking gcodes and I have a hunch I am not managing my carriage states properly but I cant check it.