Klipper3d/klipper #195: Feature Request: TMC2130 SPI communication

maz3max, 2018-02-26:
So far, klipper looks fantastic to me.
Is there a chance to control the Trinamic TMC2130 over the AVR slave board (or connected directly to the raspberry)?
That might not be a priority, but then you can have cool things like stall detection.

gdachs, 2018-02-26:
In the meantime you could try this fork: https://github.com/cruwaller/klipper/commits/lpc176x

NeelD87, 2018-02-26:
Its WIP , check the https://github.com/KevinOConnor/klipper/tree/work-tmc2130-20180217 branch

maz3max, 2018-02-26:
Thanks, guys!
I am very happy you keep this project moving :smile:

KevinOConnor, 2018-03-02:
As noted, there's been a bit of work on this.  There's also a long discussion in issue #101.  I'm going to close this issue in favor of that one.

closed 2018-03-02 by KevinOConnor

userosos, 2018-03-05:
How i can add support control TMC2130 SPI?
I did not find it in https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg .