Klipper3d/klipper #197: Installation in Windows

fakio87, 2018-02-27:
Hi,

I am running Octoprint on a Windows enviroment.

Is there any way to install Klipper under windows? Of course it fails to run the .sh file. 

However as it is basing on Phyton it should work cross plattform right? Any way to install it manually under windows unsing cmd?

kind regards
fakio

mmone, 2018-02-27:
My opinion: Do yourself a favor and get a Raspberry Pi. Even if your unfamiliar with Linux it is very easy to setup octoprint with the [octopi distribution](https://octoprint.org/download/). Plus you will no longer have to run your PC all the time when printing.

fakio87, 2018-02-27:
Thank you for your advice mate.
I have two raspberry Pi here, a zero which I set up as Wifi Cam for Octoprint and a 3b for hosting Minecraft. I know how to set them up. I used the zero as Octoprint server before and experienced a very, very slow print plus very slow access speed to the webpage. 
The printing speed must have resulted from the weak cpu of the zero, nevertheless I doubt that the 3b will give me equal access speed than the device I purchased especially for the Octoprint server (to be able to run two instances for two printers at once). I got a HIGOLE GOLE10 Mini PC for that purpose as the cpu seems capable of doing the job and it includes a display as well.
However that device is unwilling to boot any linux stick for install. All my other devices accept the boot sticks and function well - even my mac - but not the Higole. So I am stuck with windows.

That is the reason why I am asking whether there is any way to install Klipper on Windows.

mmone, 2018-02-27:
Yes the Zero is borderline underpowered with Octoprint alone, together with Klipper it definitely is. I've tried that route first too. Now I have Klipper running on a Rpi2. That setup runs perfectly smooth with a responsive octoprint. I have to add that I don't use cam streaming though. As far as windows goes I can't really help. I know that [octoprint runs on it] (https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows).
A little bit convoluted but in theory you could run Klipper host on the Zero and octoprint on your MiniPC.

fakio87, 2018-02-27:
Hi,

I think the printer can be connected to one device at a time not two, right?

I am quite happy with the performance of the current setup, if I could get Linux running on it there would be no issue here. but I am stuck with this now.

In the end I just need to know how to install all related content from the .sh files. Or maybe this can be translated somehow so that it is executable from windows cmd (like a bat file).

mmone, 2018-02-27:
> I think the printer can be connected to one device at a time not two, right?
My idea was simply put to make the connection between octoprint and the Klipper host process ,that is now done locally via a virtual comport that the klipper process provides, over TCP/IP networking. It should be possible to forward the comport with netcat from the Pi.

> In the end I just need to know how to install all related content from the .sh files.

I don't know if all dependencies can be met under Windows.

fakio87, 2018-02-27:
Hi,

Thank you for your fast responses. I understand your meaning, still this might lead to another device that can cause lags or issues. I really need to look into getting Linux running on that machine.

TunaLatte, 2018-02-27:
I don't recommend serial over IP setup as I had timer too close randomly on mine.
I simply use virtual machine on my windows NUC, vmware preferably as the USB implementation is reliable compared to virtualbox.

fakio87, 2018-02-27:
So you run a virtual instance of linux on your windows machine for octoprint or just generally?

TunaLatte, 2018-02-27:
@fakio87 yes for octoprint and klipper, I have two VM's for 2 printers.

fakio87, 2018-02-27:
Thought that would be tough for performance, will test it anyways. Thanks

dragonnn, 2018-02-27:
Windows 10? Maybe you can use linux subsystem in it for running klipper and octoprint.

TunaLatte, 2018-02-27:
FYI I'm running Intel NUC thin canyon which uses a slow atom CPU, no
stuttering on high facet circle. Klipper is generally lightweight, I also
highly recommend you use virtual SD function, as octoprint is always slow
no matter how powerful your CPU is, visible on high poly files. I assume
any x86 instance wether it's virtual or host is still much faster than any
RPi.

TunaLatte, 2018-02-27:
It's a subsystem, you can't access serial or manage services (systemd etc)

dragonnn, 2018-02-27:
> It's a subsystem, you can't access serial or manage services (systemd etc)
https://blogs.msdn.microsoft.com/wsl/2017/04/14/serial-support-on-the-windows-subsystem-for-linux/ as far I know you can access serial ports, buf I never used it.

TunaLatte, 2018-02-27:
> support-on-the-windows-subsystem-for-linux/ as far I know you can access

It's a shared and mirrored port, there is a high chance that the data could
get corrupted.

dragonnn, 2018-02-27:
Probably the same chance when forwarding serial port o USB device to VM. Without any testing you can not say.

TunaLatte, 2018-02-27:
No, forwarding is done on USB stack level not serial. :-(

doctor3lo, 2018-03-02:
Hi! 

I have intrecting rare computer, and can't install any linux on it. Only win.
Of course, I may by micro PC, but want to try Klipper on my staff.

Also can't to install Klipper in Manjaro linux.

KevinOConnor, 2018-03-02:
FYI, Klipper is targeted to run on low-cost embedded machines (like the Raspberry Pi).  The Klipper host software has some real-time requirements - these aren't stringent - usually no more than a few hundred milliseconds.  However, people tend to use general purpose machines to do general purpose activities - like playing 3d computer games, browsing the web, defragmenting their disks, etc.  These activities don't match well with Klipper's requirements.

I'm sure it would be possible to port the host code to Windows (very little is Unix specific) if someone wishes to do that.  However, I think a low-cost embedded machine (eg, a Raspberry Pi) is a worth while investment for a 3d printer regardless.  So, I don't plan to work on porting.

doctor3lo, 2018-03-02:
Anyway, thanks for great work!

I'm trying it before on old miniPC with debian.
Also there are more chances for working with a few printers.

closed 2018-03-13 by KevinOConnor

ctmakro, 2021-05-25:
I just install the whole thing on windows and got working.

https://gist.github.com/ctmakro/c6749ca20798b8e3950f5baa47e05fba