Klipper3d/klipper #201: M23 + virtual SD card doesn't handle gcode checksums

ghandye, 2018-02-28:
A minor issue, but the implementation of M23 and the virtual SD card feature doesn't filter out the trailing asterisk in gcode checksums.

I didn't include the full debug logs because this one seemed straightforward, but here's the snippets:

Snippet from Octoprint log (M105 temperature call is interleaved here):

```
Send: N20 M105*21
Send: N21 M23 /original_top.gcode*71
Recv: ok T0:175.1 /0.0 B:75.1 /0.0
Recv: !! Unable to open file
Recv: ok

```
Snippet from klippy.log:

```
virtual_sdcard file open
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 69, in cmd_M23
    fname = files_by_lower[filename.lower()]
KeyError: 'original_top.gcode*71'
Unable to open file
```

Admittedly, using gcode checksums between Octoprint and a local /tmp/printer device is a bit silly (and I've since disabled this remnant of my previous Octoprint config) but I probably won't be the only one who accidentally tries this.

KevinOConnor, 2018-03-02:
Thanks.  Should be fixed as of commit fc60bda4.

ghandye, 2018-03-07:
Looks good:

```
Send: N29 M23 /pie_chart.gcode*45
Recv: File opened:pie_chart.gcode Size:1144777
Recv: File selected
Recv: ok

```

Closing. Thanks.

closed 2018-03-07 by ghandye