Klipper3d/klipper #202: Support for G2/G3 Controlled Arc Move

ChainsDD, 2018-02-28:
[G2/G3 definition](http://reprap.org/wiki/G-code#G2_.26_G3:_Controlled_Arc_Move)

While I've not seen a slicer for printing output G2/G3, I also use my printer for laser engraving, and it's a common command when generating G-code from an SVG. Currently I have to flash Marlin back onto my printer before using my laser for that reason.

Marlin handles planning arc moves [here](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L3400) and [here](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L13414), but I haven't dug enough in the Klipper code base to figure out how to implement it here.

hg42, 2018-03-01:
look at cruwaller's fork, he has implemented G2/G3 (though I havn't used it)

ChainsDD, 2018-03-01:
You're right, [found it](https://github.com/cruwaller/klipper/commit/fc3fffda5f108007bd75614e2b34fbeacc1af344).

It looks like his implementation is directly pulled from Marlin, and doesn't seem to take advantage of the different way that klipper works, but it does give me a good starting point to work on my own. Thanks!

KevinOConnor, 2018-03-13:
Feel free to submit a "pull request" if support is added (ideally, the arc support would be in a new file in klippy/extras/).  I'm going to close this issue for now.

closed 2018-03-13 by KevinOConnor

JohnEdwa, 2019-03-16:
It's been a year, anyone willing to have a go implementing this?
I promised to plot some wedding invites for a friend in a few months and I'd rather like skipping setting up Marlin again just for it if possible :)

danielkucera, 2019-11-06:
Someone should have mentioned here that this has been recently implemented:
https://github.com/KevinOConnor/klipper/commit/7d8c70363a4241ce7c16868e347d7c0f7855f5f3#diff-71420708e69755e8a44d7e3eac23cf1b

mdelanno, 2020-05-11:
I work on a slicer that takes BRep geometry as input. When I slice a cylinder with a vertical axis, I get a circle and I can generate a G3 command. I was wondering if it would be possible to generate real arcs/circles (not a batch of linear moves but the adequate pulses to drive the motors in a circle trajectory) with Klipper and if so, would it be complicated?

jalanjarosz, 2020-05-11:
Klipper does support G2/G3 arc moves using I’s and J’s

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#g-code-arcs

Thanks!
James

mdelanno, 2020-05-11:
Yes, I know that... My question is: can a real circular path be generated instead of a succession of G1 commands?

jalanjarosz, 2020-05-11:
Current slicers read in .stl or tessellated 3D models.  They cannot create arc movements from models containing flats.

If a 3D slicer can read a solid model, native or B-Rep, and can create G2/G3 moves, than the printer can print arc movements.

At the moment, Cura, Slic3r, Pursa Slicer, etc.  cannot do this.

It’s more likely, as MCAD machining apps mature to include additive manufacturing, this will happen here first.

JohnEdwa, 2020-05-11:
That is exactly what G2 and G3 are for. If the slicer generates a G2 arc move, Klipper will execute it as a G2 arc move.

mdelanno, 2020-05-11:
My slicer (Fanplast1k) is working in AutoCAD. I use the functions provided by AutoCAD to slice the model so I get a composite curve with arcs or circles. I translate them in G3 commands (I always turn in counter clock-wise direction). 

The current operation (conversion of G2/G3 to G1) produces an acceptable result if the resolution is small, but I would like to know if we can go further and do without this resolution.

KevinOConnor, 2020-05-11:
> I was wondering if it would be possible to generate real arcs/circles (not a batch of linear moves but the adequate pulses to drive the motors in a circle trajectory) with Klipper and if so, would it be complicated?

It is possible, but it's not easy.  Klipper currently uses an "iterative solver" to find precise step times for every linear move.  It should be possible to change Klipper to find precise step times using the same "iterative solver" with non-linear moves.  However, doing this would require updating Klipper's look-ahead code to handle non-linear moves.  (See the docs/Kinematics.md and docs/Code_Overview.md files for further information.)  That would be notable work, and it is unclear if the result would be a sufficient improvement over the current G2/G3 "line segmentation" method.

FWIW, since G2/G3 commands only handle circular moves (and not other common arcs), I'm not convinced those commands have actual value.  As such, I'd add "implementing new arc commands" to the list of notable work that would need to go with the above.

Cheers,
-Kevin

Peuqui, 2020-06-13:
Hi,
this guy is developing a plugin to convert G1 into G2/G3 code to reduce code and stuttering. It seems a highly promising piece of software:
https://community.octoprint.org/t/new-plugin-anti-stutter-need-testers/18077

Regards

mrx23dot, 2021-02-02:
The gcode post-processing app is ready, it takes G1 linear moves and merges them into G2/G3 arcs (where it makes sense), it can reduce the gcode size by -90% for pipes. https://github.com/FormerLurker/ArcWelderLib/releases

`bin\ArcWelder.exe -l=INFO -p --allow-dynamic-precision --path-tolerance-percent=0.05 --resolution-mm=0.05 in.gcode out.gcode`

The goal was to also reduce shaking/stutter by not using the slicer's crude linear lines for arcs. But according to Kevin the same thing happens in Klipper. The best we can do is to turn up the resolution on Klipper end until raspberry doesn't overload.

```
[gcode_arcs]
resolution: 1.0
#   An arc will be split into segments. Each segment's length will
#   equal the resolution in mm set above. Lower values will produce a
#   finer arc, but also more work for your machine. Arcs smaller than
#   the configured value will become straight lines. The default is
#   1mm.
```