Klipper3d/klipper #203: Probe is offset from nozzle in x, y and z axis

w1ebr, 2018-02-28:
Hello Kevin,
Klipper isn't the problem, knowing how to specify the offset of the probe touch point from the nozzle location is my question.

The locations used in the bed tilt set of points are relative to the nozzle. Relative to the nozzle x,y,z coordinates, the bltouch probe point is at x-0, y-27.0, z-0.62.  Whenever the probe is used, the firmware has to take the probe offsets into account so that the bed tilt map points are translated from nozzle coordinates to probe coordinates in order to move the probe to the desired measuring location. The z offsets measured are then stored with the nozzle coordinates. 

In my case, if I want to probe the bed height at 10,10 the nozzle has to be moved to 10,37, then the probe is deployed. The z height is then measured.

The difference between the z trigger point and the tip of the nozzle can be handled in a few different ways that I know of.

How do I specify the probe point offsets?

Thanks!

KevinOConnor, 2018-03-02:
On Wed, Feb 28, 2018 at 05:52:48PM +0000, w1ebr wrote:
> Hello Kevin,
> Klipper isn't the problem, knowing how to specify the offset of the probe touch point from the nozzle location is my question.
>
> The locations used in the bed tilt set of points are [...]

If you want to probe the point where the nozzle is at 10,10, then you
would place 10,10 in the points list in the config file.  If you want
to probe the point where the probe itself is at 10,10 then you'd place
10,37 in the points list.

> The difference between the z trigger point and the tip of the nozzle can be handled in a few different ways that I know of.
> 
> How do I specify the probe point offsets?

For basic bed tilt calibration, it shouldn't be necessary to specify
the x and y probe offset.  The goal is only to determine the slope of
the bed.  The slope shouldn't change with an x/y probe offset.

That said, if you use the probe to both home the z axis and as part of
the bed_tilt_calibrate command then things get confusing.  I suspect
this is also impacting issue #176.  I don't have the hardware, so it's
hard for me to test directly.  A good log would help as I could then
walk through an example to see what the math is calculating.

-Kevin

w1ebr, 2018-03-02:
I understand, for a simple bed tilt the x and y offset don't matter. I am used to a more complex bed leveling scenario where the value at the probe points matter AND I am used to specifying things in nozzle coordinates. I am hereby making a future implementation request :-) where I can specify the x, y offset of the probe relative to a nozzle.

closed 2018-03-10 by w1ebr