Klipper3d/klipper #206: Build environment issues

JeanJacquesBluffobaque, 2018-03-01:
Not entirely sure this belongs here, but Google searches haven't provided a solution. I've been running Octoprint/Octopi for about 2 years. It's completely up to date. I've followed https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md to obtain Klipper.

At the 'Make' step, I get the error that setjmp.h is missing. I can _locate_ it, there are actually 2 instances of it. I can add the paths to the makefile using -I but that results in further missing header files.

As I've double checked that I followed all steps detailed in the Installation.md, I'm at a loss. How do I fix this?

lenne0815, 2018-03-01:
Without any further knowledge it sounds like a permission error.

KevinOConnor, 2018-03-02:
On Thu, Mar 01, 2018 at 12:41:09AM -0800, JeanJacquesBluffobaque wrote:
> Not entirely sure this belongs here, but Google searches haven't provided a solution. I've been running Octoprint/Octopi for about 2 years. It's completely up to date. I've followed [...]

If you attempted the Klipper install from an OctoPi installation that
was originally done 2 years ago, then I think you should retry after
installing OctoPi 0.14 from scratch to an sdcard.

-Kevin

JeanJacquesBluffobaque, 2018-03-03:
Install from scratch worked.

closed 2018-03-03 by JeanJacquesBluffobaque