Klipper3d/klipper #215: configure file for MKS Gen V1.4

userosos, 2018-03-03:
Who uses the motherboard? I want an config example for the motherboard.

mmone, 2018-03-03:
I'm waiting for delivery of a MKS Gen L that's on a slow boat from China :-). Their product naming scheme is somewhat confusing many seemingly identical boards with different naming.
But they have "documentation" of their boards on [github](https://github.com/makerbase-mks/Datasheet). Most boards are compatible with RAMPS if not it says so in the pdf. All docs include a configuration guide for Marlin. That should enable you to figure out the pinout with the Marlin board definition as a basis.
And Kevin wrote a [guide](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md) on howto verify your config/pinout to avoid issues.

KenshiHH, 2018-03-03:
just use the Ramps config, its compatible

userosos, 2018-03-04:
@KenshiHH This has little difference from RAMPS nedd add fan pin. I see it in Marlin firmware.

KenshiHH, 2018-03-04:
I'm using a MKS Gen 1.4 and used the Ramps one as a base config.

if you want to double check the pins, just use this picture as a reference
http://reprap.org/wiki/File:MKS_GEN-PIN.PNG

theopensourcerer, 2018-03-04:
Just set mine up today on a CoreXY. used the RAMPS config for the pins (Had to reverse direction of Y and Z IIRC) but otherwise it seems to be working fine.

jakep82, 2018-03-05:
I've also used the RAMPS config on an MKS Gen 1.4 board, and all of the pins were correct.  The only changes I made were to reverse directions on a couple steppers.  This board doesn't need a separate config file.

closed 2018-03-13 by KevinOConnor