Klipper3d/klipper #217: Negative End stop positions?

theopensourcerer, 2018-03-04:
I'm not even sure if this is a valid question but here we go.

On my CoreXY printer (HEVO) I have deliberately designed in extra space around the build platform it self.

What I tried to so today, was to set the position_endstop equal to roughly half of the available space around the bed in either the X or Y axis. The idea being I can print within a 0 to 300mm range as my slicer would understand it but for some custom code, may be a nozzle cleaning routing etc, I can go into this negative area. But Klipper didn't like me trying home an axis with a negative endstop position... Said it was out-of-range.

Am I being daft with this?

Papsnoek, 2018-03-04:
Hi @theopensourcerer , I recognize you from the Facebook Hypecube group :)

Here is my X Stepper config that I used when I ran Klipper firmware:
[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = 0.00625
endstop_pin = ^ar3
position_endstop = -70
position_min = -70
position_max = 180
homing_speed = 20

This worked perfectly for me. So unless anything changed since then with these settings they should send you on the right track

theopensourcerer, 2018-03-04:
Ahh Eureka! Many thanks for the cluebat @Papsnoek! I think I was missing changing the "position_min" as well as the position_endstop params.

Closing.

closed 2018-03-04 by theopensourcerer