Klipper3d/klipper #220: Z offset errors.

rewolff, 2018-03-07:
(Kevin: I don't know if there is another way to discuss subjects with you (and the rest of the community).... Mailing list? Forum?)

I always had places where adhesion was bad or the head would be so close to the surface that extrusion became impossible.... So I built a primitive Z-probe to at least chart the issue....

Attached is the probe log. I visualize that in gnuplot with:
splot 'probe2.txt' using 1:2:3 w linespoints

[probe2.txt](https://github.com/KevinOConnor/klipper/files/1789071/probe2.txt)

Because I didn't reset the log before doing the final run, you can get a feel for the repeatability in the middle. 

So... This is what a height offset map looks like..... 

Of course, the big slant is something I better fix with the homeing adjustments. But there is going to be some bumps left that need "fixing"....

herrwusel, 2018-03-07:
I think it's hard to tell if that is a Klipper error or something else. Do you have a map available that wasn't created by klipper?

rewolff, 2018-03-07:
I think you are misunderstanding the issue. I probably wasn't clear enough. 

Klipper was installed like 2 weeks ago, print quality improved. I don't want to go back. The Z-probe was installed 2 days ago. This is the second "probe the whole bed" result. 

I do NOT think that the errors are caused by KLIPPER. It's the mechanics, and adjustments that cause the errors. it's a DELTA printer, as you might know. I built it with a friend. He has a strong mechanical background in MAKING things. So knowing that it helps to have the moving end as light as possible, he also wants the rods to be light. So where the "howto" specifies putting a M3 rod inside the carbon tube he glued a small piece of M3 in the carbon tube. So in the "howto" you end up with rods that can differ by the M3 thread pitch, or by 0.5mm. Easy enough to measure. But now with the glued ends, there can be an arbitrary difference in the threads resulting in an arbitrary offset. Maybe that's the cause. We've of course adjusted the rods to within 0.5mm, but we can't get any closer. 

Anyway, with a lot of unknowns, the probe clearly shows that I have a tilt. I'll correct that. with the usual means. 

The main purpose of this posting is to demonstrate the need to have a per X-Y Z-offset. With adjusting the endstop screws I'll be able to get rid of the tilt, but not of the general humps. When that is structural, I'd like to have klipper be able to adjust the moves for those hills. As software gets better, you can create better and better results with hardware that has faults/shortcomings. For example, cameras are no longer manufactured with 8 or 24 million light sensor areas that all work. Or are all more-or-less equal. No: They are all different, and a calibration process will bring the results to a workable level. 

Anyway... My printer has odd deficiencies. I don't know yet what the cause is. But being able to measure them is a step towards solving them. One of the other options that might be causing this is that the GT2 pulleys on the steppers are not round, i.e. have their hole off-center. This would mean that 20mm higher (half a rotation of the pulley) the offsets would be completely different. 

Being able to correct for the currently measured offsets-near-Z=0 would improve bed adhesion so that I can print larger objects. And printing say a staircase of 50mm with 5mm steps at different locations would allow me to measure if my hypothesis of the differing stepsizes along the rotation of the stepper is correct..... 

I want to make people familiar with the code (Kevin?) aware of what a bed-leveling-probe session looks like, possibly motivating someone to implement it before I get around to it.  

Reading up on probing indicated to me that Kevin was thinking a probe would allow measuring a bed-tilt but that those would be the extent of the errors encountered. My mirror is likely MUCH flatter than what is measured here. The errors come from a different source, but can potentially be corrected for in software..... 

---------------
Measuring the roundess of the pulleys:
* extend a digital caliper all the way. 
* Move the head to near the A tower, Z=320mm. 
* Lower the head until it pushes the caliper. Note the Z and caliper measurements. 
* lower the head in 2mm increments and note the numbers. 
Find a way to better visualize the data.

cmock, 2018-03-08:
Did you issue the DELTA_CALIBRATE command and update the configuration accordingly?

I have a Kossel mini that's not particularily well-made, and I printed an adapter that puts a microswitch right below the nozzle and probed that way; then entered all the corrections in the configuration, adapted the Z height (by adding the same value to position_endstop on all steppers) iteratively until it was correct, and that's it.

I think that as long as your print surface really is flat (and there's nothing better than glass/mirrors for that IMO), you don't need a height map feature.

rewolff, 2018-03-08:
> Did you issue the DELTA_CALIBRATE command and update the configuration accordingly?
Not really. Haven't gotten around to that yet. As in I started reading the documentation and more or less stopped with that once it worked. I'll start looking at what "delta_calibrate" does and how to use it.

gdachs, 2018-03-08:
@rewolff: to you initial question: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

KevinOConnor, 2018-03-08:
On Wed, Mar 07, 2018 at 06:17:39AM -0800, Roger Wolff wrote:
> I do NOT think that the errors are caused by KLIPPER. It's the
> mechanics, and adjustments that cause the errors. it's a DELTA
> printer, as you might know.

As was mentioned by @cmock, you should run DELTA_CALIBRATE - the delta
parameters have a complex impact to the bed height that is not at all
obvious.  In general, mesh bed leveling is not a good fit for deltas.
That said, I have nothing against mesh bed leveling - it's just not
something I'm personally planning to work on in the immediate future.

[...]
> Klipper was installed like 2 weeks ago, print quality improved. I
> don't want to go back. The Z-probe was installed 2 days ago. This is
> the second "probe the whole bed" result.
[...]
> So in the "howto" you end up with rods that can differ by the M3
> thread pitch, or by 0.5mm.

FYI, arm length discrepancies can have a dramatic impact on the
accuracy of the probe (depending on how the probe is implemented).
Minor differences in the arm length will result in an axial twist to
the effector.  That twist could result in a significant change in
probe height when the effector is at different XY locations.

I goofed when I assembled the arms of my delta and as a result the
probe that came with the kit is useless.  The printer still works well
after running DELTA_CALIBRATE in manual mode though.

I recommend running DELTA_CALIBRATE at least once in manual mode
(which allows one to use the "paper test" to adjust the height at each
point and then run NEXT).  If one gets similar results in both manual
and automatic probe mode, then the probe is fine.

-Kevin

rewolff, 2018-03-08:
I calculated/measured delta-radius of 127.2 and delta_calibrate found 126.24 Is that very significant?

I do suspect a twist in the effector when moving side-to-side. Is the printer a "total loss" or would it be possible to get it to work well with a manual probe?

Why do I need to specify the Z-probe-offset? Why not go to X0 Y0 and probe it?

KevinOConnor, 2018-03-09:
On Thu, Mar 08, 2018 at 03:44:42PM -0800, Roger Wolff wrote:
> I calculated/measured delta-radius of 127.2 and delta_calibrate found 126.24 Is that very significant?

I'm not sure what you mean when you say you measured it.  The
parameters interact in non-obvious ways, so there isn't a straight
forward way to measure the delta radius with a ruler, calipers, etc..
That said, a difference of one mm doesn't sound bad.

> I do suspect a twist in the effector when moving side-to-side. Is the printer a "total loss" or would it be possible to get it to work well with a manual probe?

I'd set the delta_calibrate manual_probe=True in the config and try it
out.  It should only take a few minutes.  (Update the config to use
manual mode, run RESTART, run DELTA_CALIBRATE, and then at each point
that the head moves to, move the z axis up and down (eg, G1 Z0.8 F300)
until a piece of paper moves between the head and bed with a small
amount of friction, then run NEXT and repeat at each position.)

> Why do I need to specify the Z-probe-offset? Why not go to X0 Y0 and probe it? 

The z_probe_offset is the distance between the nozzle and bed when the
probe triggers.  If you're using manual mode, set this to zero.  If
you're using an automatic probe, then it's an attribute of the probe -
you need to know that distance.  (If you don't know the distance, go
to 0,0 and run PROBE, run M114, and note the Z height.  Then do the
"paper test" above, at the same XY position, then run M114, and the
z-probe-offset is the different in z between the two M114 reports.)

-Kevin

rewolff, 2018-03-09:
Will do! Thanks. 

> I'm not sure what you mean when you say you measured it. 

When moving to klipper I copied it over from the marlin firmware that I compiled before. It's already a long while ago that I configured THAT... Maybe we found it on the internet somewhere. (It's a "beagle" from reprapworld.com). Maybe we did get creative with calipers. :-)

rewolff, 2018-03-11:
FYI. The first printer didn't have a "delta_calibrate" command. That's how long it's been running klipper. :-) 
I've changed the parameters, and it seems to be doing a bit better than before. 

Still... I'm seeing that there is an area between the extremities of the current print where the first layer is "too high". Not in the middle, not towards one of the towers, but like halfway between the middle and the outside of the print......

When trying to manually calibrate the second printer again, to verify if it now behaved better... The head won't move below 0.0 . So any small negative offset cannot be measured. Would it be an idea to make the "box"  abit larger during "delta_calibrate" ?

From the time it takes between the last "next" and the printing of the results I deduce that some sort of iterative algorithm. is run to optimize the parameters. Right? 

With the angled almost completely fixed at 210-330-90, there are only four output parameters and seven measurements. So I would expect that from the simulation, you've got enough information to give an esitimate of the largest offset. And if that is say larger than 0.2mm, then a warning might be a good idea. 

"offset" here is: move to the exact stepper-position where the user (or probe) reported: Z=0, and then move to Z=0 with the new settings. What's the distance moved?

The second printer seems to have a loose connection to the temp sensor, and doesn't print today.

KevinOConnor, 2018-03-11:
On Sun, Mar 11, 2018 at 02:06:20PM +0000, Roger Wolff wrote:
> FYI. The first printer didn't have a "delta_calibrate" command. That's how long it's been running klipper. :-)
> I've changed the parameters, and it seems to be doing a bit better than before. [...]

Yeah - I goofed on the construction of the arms on my delta - so, I
know what you mean.  One of the things I'd like to try out is an 8
parameter optimization (endstop positions, angles, radius, arm
lengths).  To do this well, I believe it will be necessary to print a
calibration object and measure some lengths on it with calipers.

> When trying to manually calibrate the second printer again, to verify if it now behaved better... The head won't move below 0.0 . So any small negative offset cannot be measured. Would it be an idea to make the "box"  abit larger during "delta_calibrate" ?

Set the printer minimum_z_position=-1 to permit this.

> From the time it takes between the last "next" and the printing of the results I deduce that some sort of iterative algorithm. is run to optimize the parameters. Right? 

Yes.  Klipper uses Coordinate Descent for parameter optimization.

> With the angled almost completely fixed at 210-330-90, there are only four output parameters and seven measurements. So I would expect that from the simulation, you've got enough information to give [...]

You can play with the parameters optimized if you wish - see
klippy/extras/delta_calibrate.py:finalize() - the adj_params lists
which parameters will be optimized.  (If you do, be sure to run "sudo
service klipper restart" after each software change.)  The klippy log
file of your last probe will show what parameters are available.  (It
also has more detailed information on the probe points and their
results.)  In my tests, attempting other parameters does not work well
- the parameters interact with each other in non obvious ways - the
returned results provide a flat bed but result in absurd settings for
any other position on the printer.  This is why I believe further
parameter optimization would require additional measurements from a
printed calibration part.

-Kevin

closed 2018-03-18 by KevinOConnor