Klipper3d/klipper #224: install instructions beaglebone

3DSTB, 2018-03-08:
Please can you put some informations for the following setup into the docs?
I want to use Klipper in the following config:
Host: PI 3 with Klipper and Octprint
Controller: BBB with Cramps
MY Questions:
1.) "The building and flashing the micro controller" in Installation.md is not usable for this setup? Insteed of this i must "building the micro controller code" in beaglebone.md?
2.) I must do all steps of the beaglebone.md for my setup? Also this: 

For the Replicape, it is also necessary to compile and install the micro-controller code for a Linux host process. Run "make menuconfig" a second time and configure it for a "Linux process":

_make menuconfig_
Then install this micro-controller code as well:

_sudo service klipper stop
make flash
sudo service klipper start_

KevinOConnor, 2018-03-08:
On Thu, Mar 08, 2018 at 08:27:29PM +0000, 3DSTB wrote:
> Please can you put some informations for the following setup into the docs?
> I want to use Klipper in the following config:
> Host: PI 3 with Klipper and Octprint
> Controller: BBB with Cramps

Klipper has to run on the BBB.  It's technically possible to run
OctoPrint on another machine, but that's not straight forward.  I'd
definitely get everything running on the BBB first before introducing
an RPi.

I haven't had any success reports with the CRAMPS setup - mostly due
to weird interactions with the "cape management" stuff.  That said, it
should work, so let me know how it goes.

> MY Questions:
> 1.) "The building and flashing the micro controller" in Installation.md is not usable for this setup? Insteed of this i must "building the micro controller code" in beaglebone.md?

Yes - follow the instructions in beaglebone.md.

> 2.) I must do all steps of the beaglebone.md for my setup? Also this: 
> 
> For the Replicape, it is also necessary to compile and install the micro-controller code for a Linux host process. Run "make menuconfig" a second time and configure it for a "Linux process":

The CRAMPS should not require the "linux process" step.  (On the
replicape, we need to access the replicape specific PWM and
shiftregister chips via the "linux process".)

That said, another user on CRAMPS had slightly better luck configuring
the analog inputs via the "linux process" - if you run into problems
with the example cramps config, let me know.

-Kevin

3DSTB, 2018-03-08:
Thank you for the replay. Is this something for the "cape menegment"?
https://github.com/hzeller/beagleg/blob/master/hardware/CRAMPS/BeagleG-CRAMPS.dts

3DSTB, 2018-03-10:
I use now
cap universal enabled in uEnv.txt at /boot and config the pins with the config-pin command.

config-pin P8.13 gpio_pd ...

3DSTB, 2018-03-10:
i made a script to load the pin settings at startup:
#!/bin/bash
		      # P9.9   Power Button
		      # P9.10  Reset Button
config-pin P9.23 out- # GPIO1_17, output, pulldown, machine_pwr
config-pin P9.14 out+ # GPIO1_18, output, pullup, all_stepper, enable_pin
config-pin P8.26 out- # GPIO1_29, output, pulldown, estop_sw
config-pin P8.17 in   # GPIO0_27, input, estop
config-pin P9.25 out- # GPIO3_21, output, pulldown, led
config-pin P8.13 out- # GPIO0_23, output, pulldown, stepper_x, step_pin
config-pin P8.15 out- # GPIO1_15, output, pulldown, stepper_y, step_pin
config-pin P8.19 out- # GPIO0_22, output, pulldown, stepper_z, step_pin
config-pin P9.16 out- # GPIO1_19, output, pulldown, extruder(0), step_pin
config-pin P9.17 out- # GPIO0_5, output, pulldown, extruder(1), step_pin
config-pin P9.24 out- # GPIO0_15, output, pulldown, extruder(2), step_pin
config-pin P8.12 out- # GPIO1_12, output, pulldown, stepper_x, dir_pin
config-pin P8.14 out- # GPIO0_26, output, pulldown, stepper_y, dir_pin
config-pin P8.18 out- # GPIO2_1, output, pulldown, stepper_z, dir_pin
config-pin P9.12 out- # GPIO1_28, output, pulldown, extruder(0), dir_pin
config-pin P9.18 out- # GPIO0_4, output, pulldown, extruder(1), dir_pin
config-pin P9.26 out- # GPIO0_14, output, pulldown, extruder(2), dir_pin
config-pin P8.8 in+   # GPIO2_3, input, pullup, stepper_x, endstop_pin (min)
config-pin P8.7 in+   # GPIO2_2, input, pullup, stepper_x, endstop_pin (max)
config-pin P8.10 in+  # GPIO2_4, input, pullup, stepper_y, endstop_pin (min)
config-pin P8.9 in+   # GPIO2_5, input, pullup, stepper_y, endstop_pin (max)
config-pin P9.13 in+  # GPIO0_31, input, pullup, stepper_z, endstop_pin (min)
config-pin P9.11 in+  # GPIO0_30, input, pullup, stepper_z, endstop_pin (max)
config-pin P8.11 out- # GPIO1_13, output, pulldown, heat_bed, heater_pin
config-pin P9.15 out- # GPIO1_16, output, pulldown, extruder(0), heater_pin, may use pwm insteed of out-
config-pin P9.27 out- # GPIO3_19, output, pulldown, extruder(1), heater_pin, may use pwm insteed of out-
config-pin P9.21 out- # GPIO0_3, output, pulldown, extruder(2), heater_pin, may use pwm insteed of out-
config-pin P9.41 out- # GPIO0_20, output, pulldown,  fan1, may use pwm insteed of out-
config-pin P9.22 out- # GPIO0_2, output, pulldown, fan2, may use pwm insteed of out-
                      # P9.20, GPIO0_12, sda
                      # P9.19, GPIO0_13, scl
                      # P9.31, GPIO3_14, sck
                      # P9.29, GPIO3_15, miso
                      # P9.30, GPIO3_16, mosi
                      # P9.28, GPIO3_17, spi_cs0
                      # P9.33, AIN4, head_bed sensor_pin (thermistor 0)
                      # P9.35, AIN6, thermistor 2
                      # P9.36, AIN5, extruder(0) sensor_pin (thermistor 1)
                      # P9.37, AIN2, anlaog display header
                      # P9.38, AIN3, analog display header (thermistor 3 with jp501 closed)
                      # P9.39, AIN0, analog display header
                      # P9.40, AIN1, analog display header

i put this here
sudo vi /etc/init.d/pinset.sh
sudo chmod 755 /etc/init.d/pinset.sh
sudo update-rc.d pinset.sh defaults

3DSTB, 2018-03-10:
Setting the pin values works.
One thing is still not clear: What is the term "Value" with 0 or 1? 

This is to find in the config-pin description:
hi | high | 1 :             Set pinmux to gpio and set gpio direction to output driving high
lo | low | 0 :            Set pinmux to gpio and set gpio direction to output driving low

KevinOConnor, 2018-03-14:
So, to summarize, after running config-pin and setting the pullup_resistor=2000 are you able to successfully print?

3DSTB, 2018-03-14:
I can not say this.
I'm still searching for the right pin configs. I found somes sources. But they does not work for me. But it looks like i can control the stepper, the heatbed, the endstops. I have still the problem , that the Extruder heater does not stop to heat, eg if set to 50°C he goes up to 100 and so on. 
About the behavior in a printer i can not say anything now. I'm still building the printer.

Finally i will give a short step by step description:
1.) get Image:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_iot
Sbnapshoot from 06.03.2018 used here

Install on a 4GB or bigger SD Card

1.) press Boot select Button if not disabled boot from emmc (also every following step)

2.) log in via ssh (also every following step) and change uEnv.txt
cd /boot
sudo nano uEnv.txt 
load cap universal, pru
unload video, audio
reboot

3.) get /dev/rpmsg_pru30 ready

follow this:
https://gist.github.com/alexanderhiam/2c4187c710b2c409d8dde8c4015fe007

sudo rmmod -f pru_rproc
sudo modprobe pru_rproc
rpmsg_pru30 should now be in /dev now
reboot

4.) check /dev/ for rpmsg_pru30
if not resent
sudo rmmod -f pru_rproc
sudo modprobe pru_rproc
rpmsg_pru30 should be in /dev now
dmesg | grep pru
cat /dev/rpmsg_pru30
press Strg+C to stop the output
reboot

5.) if rpmsg_pru30 is not present
cd pru-software-support-package/examples/am335x/PRU_RPMsg_Echo_Interrupt1
make
sudo cp gen/PRU_RPMsg_Echo_Interrupt1.out /lib/firmware/am335x-pru1-fw
sudo rmmod -f pru_rproc
sudo modprobe pru_rproc
rpmsg_pru30 and rpmsg_pru31 should be in /dev now
dmesg | grep pru
cd /dev
dir

Reboot if not present:
may rpmsg_pru30 will generate with a call:

echo "Let's do more with hardware!" > /dev/rpmsg_pru30
cat /dev/rpmsg_pru30

sudo vi /usr/bin/enable-cramps-pins.sh
<
#!/bin/bash
#config-pin P8.16 high
config-pin P9.23 hi #ok
config-pin P8.26 high
config-pin P8.17 in
config-pin P9.25 low
config-pin P9.14 high

config-pin P8.13 hi #?
config-pin P8.15 low
config-pin P8.19 low
config-pin P9.16 low
config-pin P9.17 low
config-pin P9.24 low

config-pin P8.12 hi #?
config-pin P8.14 low
config-pin P8.18 low
config-pin P9.12 low
config-pin P9.18 low
config-pin P9.26 low

config-pin P8.08 lo+ #ok
config-pin P8.07 lo+ #ok
config-pin P8.10 lo+ #ok
config-pin P8.09 lo+ #ok
config-pin P9.13 lo+ #ok
config-pin P9.11 lo+ #ok

config-pin P8.11 low
config-pin P9.15 low
config-pin P9.27 low
config-pin P9.21 low
config-pin P9.41
#config-pin P9.91 in
config-pin P9.22 low
#config-pin P8.22 low
#config-pin P8.23 low
#config-pin P8.24 low
#config-pin P8.25 low
config-pin P9.42 low
#config-pin P9.92 in
config-pin P9.19 low
config-pin P9.20 low
config-pin P9.28 low
config-pin P9.29 low
config-pin P9.30 low
config-pin P9.31 low
>

sudo chmod 755 /usr/bin/enable-cramps-pins.sh

sudo vi /lib/systemd/system/enable-cramps-pins.service
<
[Unit]
Description=Enable Cramps pins
After=generic-board-startup.service

[Service]
Type=simple
ExecStart=/usr/bin/enable-cramps-pins.sh

[Install]
WantedBy=multi-user.target
>

sudo systemctl daemon-reload 
sudo systemctl enable enable-cramps-pins.service

Reboot and test:
sudo systemctl status enable-cramps-pins.service

3DSTB, 2018-03-15:
What would be the correct serial to connect to the Beaglebone from octoprint running on Raspsberry?
Starting serial connect
Unable to open port: [Errno 2] No such file or directory: '/dev/rpmsg_pru30'

I have connect the Raspi via USB to the BBB.

In the Terminal Windows i can see this:
Connecting to: /tmp/printer
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6d85e610, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 115200
Recv: // Printer is not ready
Send: N0 M110 N0*125
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Changing monitoring state from 'Detecting baudrate' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Send: N2 M21*18
Recv: ok FIRMWARE_VERSION:v0.5.0-209-gef820d9 FIRMWARE_NAME:Klipper
Recv: ok
Send: M105
Recv: ok T:0
Changing monitoring state from 'Operational' to 'Offline'
Connecting to: /tmp/printer
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6d8606b0, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 115200
Connection closed, closing down monitor
Send: N0 M110 N0*125
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Changing monitoring state from 'Detecting baudrate' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Send: N2 M21*18
Recv: ok FIRMWARE_VERSION:v0.5.0-209-gef820d9 FIRMWARE_NAME:Klipper
Recv: ok
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Changing monitoring state from 'Operational' to 'Offline'
Connection closed,

KevinOConnor, 2018-03-17:
Unfortunately, there is no simple way to send input to Klipper from a remote machine.  Right now, it's best to run OctoPrint on the same machine as Klipper.

KevinOConnor, 2018-03-31:
So, were you able to get the printer working on CRAMPS with Klipper?

3DSTB, 2018-04-04:
Unfortunately I did not get any further with the setup. I will close the issue now and I will post my further experiences here when there is some progress.

closed 2018-04-04 by 3DSTB