Klipper3d/klipper #226: Unable to open port: [Errno 2] No such file or directory: '/dev/rpmsg_pru30'

3DSTB, 2018-03-10:
found this in the klipper.log

ls /dev/ | grep pru

gives nothing back. 
I use the latest Iot Image.
bone-debian-8.10-iot-armhf-2018-03-06-4gb.img.xz

3DSTB, 2018-03-10:
may i found the solution:
# rmmod pru_rproc
# modprobe pru_rproc
# dmesg | grep pru

closed 2018-03-12 by 3DSTB

manbehindthemadness, 2020-01-17:
So here is an interesting take on this problem, we have been converting our default firmware image to use klipper over at thing-printer for the Replicape boards (https://github.com/goeland86/ReFactor). After image creation everything is peachy with the exception of extremely long boot times. So to diagnose I use systemd-analyze blame and the problematic service are:
6min 32.053s networking.service
6min 31.707s generic-board-startup.service
The universal solution to preventing the networking.service from taking so long is to edit /etc/systemd/system/network-online.target.wants/networking.service and alter the timeout from 5min to 30s. Once this change has been made; however, the error you mentioned above manifests (Unable to open port: [Errno 2] No such file or directory: '/dev/rpmsg_pru30'). I have confirmed this on both a production printer and a development unit. I think what we have here is a service startup timing/order of operations problem.

BlackStump, 2020-01-18:
out of curiousity I ran 
debian@beaglebone:~$ systemd-analyze
Startup finished in 7.947s (kernel) + 22.411s (userspace) = 30.359s
debian@beaglebone:~$ systemd-analyze blame
         15.967s generic-board-startup.service
         14.082s dev-mmcblk0p1.device
          2.697s systemd-udev-trigger.service
          1.967s keyboard-setup.service
          1.755s networking.service
          1.418s klipper_pru.service
          1.216s ssh.service
          1.184s systemd-journald.service
           945ms dnsmasq.service
           801ms systemd-logind.service
           682ms rsyslog.service
           624ms connman.service
           604ms systemd-timesyncd.service
           545ms bb-wl18xx-wlan0.service
           535ms udhcpd.service
           512ms systemd-modules-load.service
           472ms kmod-static-nodes.service
           453ms systemd-remount-fs.service
           450ms systemd-udevd.service
           435ms systemd-fsck-root.service
           431ms systemd-tmpfiles-setup.service
           421ms systemd-tmpfiles-setup-dev.service
           391ms sys-kernel-debug.mount
           354ms wpa_supplicant.service
           325ms sys-fs-fuse-connections.mount
           325ms systemd-sysctl.service
           309ms dev-mqueue.mount
           293ms systemd-journal-flush.service
           292ms systemd-update-utmp-runlevel.service
           247ms user@1000.service
           245ms systemd-update-utmp.service
           240ms sys-kernel-config.mount
           216ms systemd-user-sessions.service
           176ms console-setup.service
           150ms systemd-random-seed.service
           104ms bb-wl18xx-bluetooth.service
lines 14-36/36 (END)
not sure what going on with yours but my times are a lot different

manbehindthemadness, 2020-01-18:
I think what is interesting is if I wait for ten mins or so and restart the klipper_pru service the error goes away, I am trying to isolate where it's coming from, what's not loaded that could cause the issue?