Klipper3d/klipper #227: Z offset problems with bed tilt and z probe

w1ebr, 2018-03-10:
Hi Kevin,

I am getting a move out of range error ( Z < 0 ) using bed tilt calibrate and a z probe which has a nozzle offset.

Before I attach the log file, I would like to understand how the bed tilt values work, and what the assumptions are about the probe point so I can fix my problem If possible).

My guess is that the x and y values are slopes and that maybe the z offset is the probe distance correction between the probe height specified and the calculated probe height at x=0, y=0 (or is it the intercept at x=0, y=0?)?

I begin with x=0, y=0 and the probe to nozzle offset of 0.62mm. When I run bed tilt calibrate, it comes up with x=0.000089, y=-0.002044 and z offset=0.077.

Now I know that the nozzle to probe distance is not 0.077 mm. How am I supposed to specify values in config.prt? Do I put 0.077 in probe z offset in the bed tilt section and 0.62mm in z position endstop ?

Gene

KevinOConnor, 2018-03-11:
On Sat, Mar 10, 2018 at 02:14:45PM -0800, w1ebr wrote:
> I am getting a move out of range error ( Z < 0 ) using bed tilt calibrate and a z probe which has a nozzle offset.

If your probe_z_offset is very small, then you may have to allow for
negative z moves (eg, set stepper_z min_position=-5) to allow the
probe to succeed.

> Before I attach the log file, I would like to understand how the bed tilt values work, and what the assumptions are about the probe point so I can fix my problem If possible).

The log helps answer my questions about your setup (config, mcu type,
host type, software version, any errors raised, etc.), which helps me
answer your questions.  Not providing the log causes me to consume
extra time.  :-(

> My guess is that the x and y values are slopes and that maybe the z offset is the probe distance correction between the probe height specified and the calculated probe height at x=0, y=0 (or is it the intercept at x=0, y=0?)?

Yes.

> I begin with x=0, y=0 and the probe to nozzle offset of 0.62mm. When I run bed tilt calibrate, it comes up with x=0.000089, y=-0.002044 and z offset=0.077.
>
> Now I know that the nozzle to probe [...]

As noted in issue #203 and issue #176 I suspect things aren't working
correctly when the probe is used for both endstop homing and for bed
tilt calibration.  In particular, I suspect the code is determining
the correct slope of the bed, but it is not taking into account where
the slope interesects with the XY coordinates.  In particular, if the
reported z offset was 15 microns greater than the z offset actually
is, I suspect it is because your Z homing position is at a point that
is 15 microns along the tilted bed..

I suspect the following changes need to be made to the bed tilt code:
- make sure at least 3 points are provided
- remove the z_probe_offset parameter from the config
- when doing a home with a probe:z_virtual_endstop, record the XY
  position of the last home.  Use that position in the
  bed_tilt_calibrate calculations to account for greater/less Z if the
  bed has a tilt.
- separately, on a homing timeout, the code should suggest that
  stepper_z min_position needs to be updated.

Not sure when I'll be able to look at the above.

-Kevin

w1ebr, 2018-03-11:
Thanks, Kevin. I will always supply a log in the future.

KevinOConnor, 2018-03-13:
Hi Gene,

Can you try bed_tilt_calibrate with the new work-bedtilt-20180312 branch?  I added some code to attempt to compensate for the probe:z_virtual_endstop XY offsets.  I don't have a good way to test this, so let me know how it works (success or failure).

cd ~/klipper ; git fetch ; git checkout origin/work-bedtilt-20180312 ; sudo service klipper restart

-Kevin

w1ebr, 2018-03-13:
Sure, I can try it tonight (EDT).

The current code works when I set the Z probe offset ( in the z stepper section) as the actual offset, the bed tilt calculated numbers in that section and set z's minimum position to a small negative number. The reported z height does go negative as the nozzle moves toward the back of the bed on the first layer because the bed is tilted enough that the nozzle absolute height (system frame of reference) has to be changed to stay at the same height above the bed (relative height above the bed frame of reference).

What numbers do you want me to use in printer.cfg?

-Gene

KevinOConnor, 2018-03-13:
On Tue, Mar 13, 2018 at 10:57:36AM +0000, w1ebr wrote:
> Sure, I can try it tonight (EDT).
>
> The current code works when I set the Z probe offset ( in the z stepper section) as the actual offset, the bed tilt calculated numbers in that section and set z's [...]

I'd run with the same numbers (probe_z_offset set to the distance
between nozzle and bed when the probe triggers).  After running
bed_tilt_calibrate the new code should (assuming I got it right)
recommend a stepper_z position_endstop that is appropriate for your
tilted bed.

Thanks,
-Kevin

w1ebr, 2018-03-14:
Hi Kevin,
I set up printer.cfg with same probe offset numbers, ran bed_tilt_calibrate and got very reasonable results ( a very small correction number for the offset and bed tilt numbers that make sense ).  So I think it works as you intended 👍 
Thanks!

KevinOConnor, 2018-03-14:
On Wed, Mar 14, 2018 at 12:41:19AM +0000, w1ebr wrote:
> Hi Kevin,
> I set up printer.cfg with same probe offset numbers, ran bed_tilt_calibrate and got very reasonable results ( a very small correction number for the offset and bed tilt numbers that make sense ).  So I think it works as you intended 👍 

Great!  Just for completeness, could you run a test where you reset
the position_endstop, rerun bed_tilt_calibrate, and see if it manages
to report a correct position_endstop?

Thanks,
-Kevin

w1ebr, 2018-03-14:
Hi Kevin,
It didn't like postion_endstop set to 0.0. Log attached.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/1812302/klippy.zip)

KevinOConnor, 2018-03-14:
On Wed, Mar 14, 2018 at 05:43:45PM +0000, w1ebr wrote:
> Hi Kevin,
> It didn't like postion_endstop set to 0.0. Log attached.

Okay, thanks.  The probe z_position field was a bit confusing, so I
removed it in favor of auto-detecting the minimum z position for the z
axis.  Can you pull the new code and retest?

cd ~/klipper ; git fetch ; git checkout origin/work-bedtilt-20180312 ; sudo service klipper restart

-Kevin

w1ebr, 2018-03-15:
I will do it, but now I am a bit confused. Where do I specify the height distance between where the endstop occurs and where the nozzle is when the endstop occurs?

w1ebr, 2018-03-15:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/1813760/klippy.zip)
OK, here it is.

What do I do with the -0.275529 at 100,100 information ?

Thanks, Kevin.
Gene

KevinOConnor, 2018-03-15:
On Wed, Mar 14, 2018 at 07:06:55PM -0700, w1ebr wrote:
> What do I do with the -0.275529 at 100,100 information ?

I was hoping that if "[bed_tilt] probe_z_offset" is set to the
distance between the nozzle and bed when the probe triggers, then when
running bed_tilt_calibrate it would recommend a sane x_adjust,
y_adjust, and a sane "[stepper_z] position_endstop".  That is, I was
hoping that the bed_tilt_calibrate command would recommend the proper
position_endstop even when position_endstop isn't accurate at the
start of the command.

BTW, what is the stepper_z position_endstop that is correct for your
setup?  Is "x_adjust = 0.000227", "y_adjust = -0.002026", and
"probe_z_offset = 0.550906" good values for the other three settings?

-Kevin

w1ebr, 2018-03-15:
Hi Kevin,

The actual difference in height between when the z endstop triggers and the tip of the nozzle currently installed is about 0.62 mm. 

The x_adjust and y_adjust values did a good job handling the overall tilt of the glass bed. The probe Z offset value was close enough to allow printing, but was probably too small.

What makes finding offset numbers for the firmware confusing for me is that it seems that I have to switch from thinking about a single fixed difference in height between the endstop trigger and the tip of the nozzle (which doesn't change regardless of the x y coordinates) to a value that represents the nozzle height over a tilted bed at a particular coordinate.

This is my suggestion: 

Have the user specify the difference in height between the endstop trigger and the tip of the nozzle. (Ideally, this would be done in a way that would allow installing a different nozzle with a different offset easily). 

Since the endstop trigger can't be lower than the nozzle, the difference will always be >= 0, which means that bed tilt can be calculated without knowing the difference in height between the endstop and the tip of the nozzle (the z intercept value gets calculated later on).

The slopes and the probe trigger to nozzle offset distance are all that need to be specified in printer.cfg. 

At the beginning of print time, using the stored bed tilt data, the firmware calculates the difference between an imaginary level bed and the actual tilted bed at the x,y homing location and then wait for a full homing operation (G28). When the firmware homes the z axis, it applies the bed height correction factor for the homing location to the always >= 0 offset distance between the endstop trigger and the tip of the nozzle. With this data point (nozzle to glass distance at the homing location) and the x,y slopes the firmware calculates the intercept value.

What do you think?

KevinOConnor, 2018-03-15:
On Thu, Mar 15, 2018 at 12:08:01PM +0000, w1ebr wrote:
> Hi Kevin,
>
> The actual difference in height between when the z endstop triggers and the tip of the nozzle currently installed is about 0.62 mm.
>
> The x_adjust and y_adjust values did a good job [...]

I'm confused.  The code does not recommend a probe_z_offset.  The code
on the work-bedtilt-20180312 branch recommends a new x_adjust, new
y_adjust, and new stepper_z position_endstop.  Did the new code not
work?

If it doesn't work, can you grab the latest code on the
work-bedtilt-20180312, set the probe_z_offset to 0.62 (distance
between nozzle and trigger), run the bed_tilt_calibrate command again,
and forward the log so I can try to figure out what went wrong.

-Kevin

w1ebr, 2018-03-15:
Will do later tonight

KevinOConnor, 2018-03-17:
Thanks!  I committed this to the master branch.  I also made a further change - the z_offset is now specified in the [probe] config section (instead of [bed_tilt] and/or [delta_calibrate] sections).

FYI, the lcd screen shows the position of the printer wrt the config position_min/max settings (ie, the position relative to the endstops).  It does not show the position that the g-code is currently using (which is what m114 reports).  The new GET_POSITION command will show the position of the printer in the various ways they are tracked internally.

closed 2018-03-26 by KevinOConnor