Klipper3d/klipper #228: Delta Calibration: Failed to home probe: Timeout during endstop homing

kevintong116, 2018-03-11:
When I try to do a delta calibration
the printer stops after few probes and shows:
Failed to home probe: Timeout during endstop homing
What is the problem?
[klippy.log](https://github.com/KevinOConnor/klipper/files/1800250/klippy.log)

kevintong116, 2018-03-11:
I found the problem.
The z height is set too high that when the head move to Z0, the probe cannot be triggered.
Setting the probe z position to a negative value or reduce the position_endstop value solve the problem.

KevinOConnor, 2018-03-13:
Commit 45afa045 adds some hints to the error message to make this a little more clear.

closed 2018-03-13 by KevinOConnor

VanKurt, 2019-05-24:
Which value exactly needs to be set? Is it [stepper_z] -> position_min?