Klipper3d/klipper #229: verify_heater: does not detect overtemperature

cmock, 2018-03-11:
Thanks for the heater_verify code, that's an important feature.

However, as it stands now, it only takes into account failure to heat up. As soon as the temperature is above target - hysteresis, it is happy.

That leaves out the failure scenario where the thermistor works correctly, but the heater is "stuck on" (and given that MOSFETs tend to fail that way, that is a realistic scenario).

In thinking about how to implement this, I see a design decision to be made: do we want the behaviour to be symmetrical?

If yes, we'd probably need to define a range of possible ambient temperatures (to set the range for target == 0, because obviously target +- hystersis wont cut it), and take into account the asymmetrical time constants.

If no, I think a simple "target has been set lower, does the temperature move down at all" would suffice for most cases; we could keep the existing time constants if we only look at the first temperature drop.

That would, however, ignore the case where you heat up, cool down, temperature starts to drop enough to make the code happy, but then the heater turns on again and temperature rises again.

KevinOConnor, 2018-03-11:
On Sun, Mar 11, 2018 at 04:56:48AM -0700, christian mock wrote:
> Thanks for the heater_verify code, that's an important feature.
>
> However, as it stands now, it only takes into account failure to heat up. As soon as the temperature is above target - hysteresis, it [...]

Alas, there are several difficulties to doing the above:
- cooling times are dramatically different between heaters, so there
  is no good default
- it's unclear if additional logic would be an improvement to the
  existing max_temp logic
- if a problem is detected, there's little we can do, because the
  mosfet is stuck on
- I'm not aware of any firmware that implements the above (so there is
  little previous experience to help shape new code)

> In thinking about how to implement this, I see a design decision to be made: do we want the behaviour to be symmetrical?

Heating behavior is dramatically different from cooling behavior, so I
don't see how range checks could be symetrical.

> If yes, we'd probably need to define a range of possible ambient temperatures (to set the range for target == 0, because obviously target +- hystersis wont cut it), and take into account the asymmetrical time constants.

FYI, I spent a bunch of time writing code to calibrate a "first order
plus delay time" model for the heaters.  This calibration did a
remarkably good job of predicting the behavior of all of my heaters
during unit tests.  Unfortunately, the model wasn't predictive during
actual prints (due to filament extrusion and fan behavior).
Ultimately, I concluded that it was better to get a simple, proven,
heater sanity check into the code then spend more time researching a
more advanced solution.

Cheers,
-Kevin

cmock, 2018-03-11:
>  it's unclear if additional logic would be an improvement to the existing max_temp logic

That is probably the most important point, and, given the right circumstances, is enough of a failsafe.

But max_temp has to be set to a temperature that the heater can actually reach -- I'm thinking of printer reviews that say the bed never even gets to 50°C, or my own config that has an optimistic 110° configured in case I ever want to print ABS -- I never tested when the bed maxes out. So the docs should make it clear that max_temp has to be a value that can be reached for safety.

>  if a problem is detected, there's little we can do, because the mosfet is stuck on

I have my ATX power supply configured with ``shutdown_value: 0``, which turns off the 12V rail and thereby the heater.

KevinOConnor, 2018-03-12:
On Sun, Mar 11, 2018 at 05:55:59PM +0000, christian mock wrote:
> That is probably the most important point, and, given the right circumstances, is enough of a failsafe.
>
> But max_temp has to be set to a temperature that the heater can actually reach -- I'm [...]

Okay, I reworded the documentation (commit 3a757487).  I think there
is a fine line between scaring users, giving users a false sense of
security, and just overwhelming them with details.  I'm not sure the
best way to approach that.

> >  if a problem is detected, there's little we can do, because the mosfet is stuck on
> 
> I have my ATX power supply configured with ``shutdown_value: 0``, which turns off the 12V rail and thereby the heater.

Okay, but be aware that isn't a common setup.

-Kevin

closed 2018-03-22 by KevinOConnor