Klipper3d/klipper #230: Heater extruder failsafe

JeanJacquesBluffobaque, 2018-03-11:
Yesterday I was able to print calibration stuff, today they fail with 'Heater extruder failsafe' after 2 or 3 layers.

[klippy.log](https://github.com/KevinOConnor/klipper/files/1800613/klippy.log)

KevinOConnor, 2018-03-11:
On Sun, Mar 11, 2018 at 05:07:15PM +0000, JeanJacquesBluffobaque wrote:
> Yesterday I was able to print calibration stuff, today they fail with 'Heater extruder failsafe' after 2 or 3 layers.

Okay, thanks.  I think the defaults for the new heater verification
code may be too strict.  I changed the default from no more than 4
degrees out of range for 10 seconds to no more than 10 degrees out of
range for 15 seconds.

-Kevin

JeanJacquesBluffobaque, 2018-03-11:
Thx. Still too strict for my setup it seems.

[klippy.log](https://github.com/KevinOConnor/klipper/files/1800723/klippy.log)

KevinOConnor, 2018-03-11:
On Sun, Mar 11, 2018 at 06:37:15PM +0000, JeanJacquesBluffobaque wrote:
> Thx. Still too strict for my setup it seems.

Okay, thanks.

Your extruder temperature is highly unstable.  So, I don't think
there's anything wrong with raising an error given the received input
(temp noticeably below target for 35 seconds; extruder full on for
approx last 14 seconds).

If you want to continue anyway, you can add a [verify_heater extruder]
section to your config and increase the hysteresis and/or check_time
fields.  See the docs/example-extras.cfg for the details.

-Kevin

KevinOConnor, 2018-03-12:
FYI, I decided to make an additional change to the heater verification.  The code will now allow a longer time out of range if the temperature difference is small, and a shorter time if the temperature difference is large.  I'm not sure if this will help with your particular issue (you can always change the defaults if it doesn't).

JeanJacquesBluffobaque, 2018-03-15:
My temps still aren't very stable, but that seems unrelated to Klipper. I can print now.

closed 2018-03-15 by JeanJacquesBluffobaque