Klipper3d/klipper #231: Recv: // Unable to translate pin name: config_end_stop oid=6 pin=P8.8 pull_up=1 stepper_count=1

3DSTB, 2018-03-12:
Starting Klippy...
Args: ['/home/debian/klipper/klippy/klippy.py', '/home/debian/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.5.0-200-g57d342b'
CPU: 1 core ARMv7 Processor rev 2 (v7l)
Python: '2.7.9 (default, Aug 13 2016, 17:56:53) \n[GCC 4.9.2]'
Start printer at Mon Mar 12 02:02:21 2018 (1520820141.2 5335.7)
===== Config file =====
[stepper_a]
step_pin = P8_13
dir_pin = P8_12
enable_pin = !P9_14
step_distance = .01
endstop_pin = ^P8.8
homing_speed = 50
position_endstop = 297.05
arm_length = 333.0

[stepper_b]
step_pin = P8_15
dir_pin = P8_14
enable_pin = !P9_14
step_distance = .01
endstop_pin = ^P8_10

[stepper_c]
step_pin = P8_19
dir_pin = P8_18
enable_pin = !P9_14
step_distance = .01
endstop_pin = ^P9_13

[extruder]
step_pin = P9_16
dir_pin = P9_12
enable_pin = !P9_14
step_distance = .0022
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = P9_15
sensor_type = ATC Semitec 104GT-2
sensor_pin = P9_36
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = P8_11
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = P9_41

[mcu]
serial = /dev/rpmsg_pru30
pin_map = beaglebone

[printer]
kinematics = delta
max_velocity = 300
max_accel = 3000
max_z_velocity = 150
delta_radius = 174.75

[delta_calibrate]
radius = 50

=======================
Delta max build height 297.05mm (radius tapered above 247.51mm)
Delta max build radius 157.10mm (moves slowed past 141.17mm and 153.72mm)
Starting serial connect
Loaded 46 commands (v0.5.0-200-g57d342b-20180312_010648-beaglebone / gcc: (GCC) 8.0.0 20170530 (experimental) binutils: (GNU Binutils) 2.28.51.20170530)
MCU config: CLOCK_FREQ=200000000 MCU=pru ADC_MAX=4095 SOFT_PWM_MAX=256 STATS_SUMSQ_BASE=256
got {u'count': 93, '#receive_time': 5336.135487344, u'sum': 31611, u'sumsq': 62503, '#name': u'stats', '#sent_time': 5336.082657897}
Config error
Traceback (most recent call last):
  File "/home/debian/klipper/klippy/klippy.py", line 239, in _connect
    cb('connect')
  File "/home/debian/klipper/klippy/mcu.py", line 745, in printer_state
    self._connect()
  File "/home/debian/klipper/klippy/mcu.py", line 602, in _connect
    self._build_config()
    self._build_config()
  File "/home/debian/klipper/klippy/mcu.py", line 523, in _build_config
    self._config_cmds[i] = pin_resolver.update_command(cmd)
  File "/home/debian/klipper/klippy/pins.py", line 180, in update_command
    return re_pin.sub(pin_fixup, cmd)
  File "/home/debian/klipper/klippy/pins.py", line 173, in pin_fixup
    raise error("Unable to translate pin name: %s" % (cmd,))
error: Unable to translate pin name: config_end_stop oid=6 pin=P8.8 pull_up=1 stepper_count=1
Unable to translate pin name: config_end_stop oid=6 pin=P8.8 pull_up=1 stepper_count=1
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Unable to translate pin name: config_end_stop oid=6 pin=P8.8 pull_up=1 stepper_count=1
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Unable to translate pin name: config_end_stop oid=6 pin=P8.8 pull_up=1 stepper_count=1
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

hg42, 2018-03-12:
it's exactly what the error message tells you:

do you see the difference?
enable_pin = !P9_14
endstop_pin = ^P8.8

3DSTB, 2018-03-12:
I have fixed it. But i can't get the steppers to run or the fans on. The temperature signal are there.

Here is the log:
Starting Klippy...
Args: ['/home/debian/klipper/klippy/klippy.py', '/home/debian/printer.cfg', '-l$
Git version: 'v0.5.0-200-g57d342b'
CPU: 1 core ARMv7 Processor rev 2 (v7l)
Python: '2.7.9 (default, Aug 13 2016, 17:56:53) \n[GCC 4.9.2]'
Start printer at Mon Mar 12 09:19:56 2018 (1520846396.9 1394.1)
===== Config file =====
[stepper_x]
step_pin = P8_13
dir_pin = P8_12
enable_pin = !P9_14
step_distance = .0125
endstop_pin = ^!P8_8
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_y]
step_pin = P8_15
enable_pin = !P9_14
step_distance = .0125
endstop_pin = ^!P8_10
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_z]
step_pin = P8_19
dir_pin = P8_18
enable_pin = !P9_14
step_distance = .0025
endstop_pin = ^!P9_13
position_endstop = 0
position_max = 200

[extruder]
step_pin = P9_16
dir_pin = P9_12
enable_pin = !P9_14
step_distance = .0025
endstop_pin = ^!P9_13
position_endstop = 0
position_max = 200

[extruder]
step_pin = P9_16
dir_pin = P9_12
enable_pin = !P9_14
step_distance = .002
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = P9_15
sensor_type = ATC Semitec 104GT-2
sensor_pin = P9_36
control = pid
pid_kp = 27.7
pid_ki = 3.04
step_distance = .002
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = P9_15
sensor_type = ATC Semitec 104GT-2
sensor_pin = P9_36
control = pid
pid_kp = 27.7
pid_ki = 3.04
pid_kd = 62.98
[... 63 lines cut ...]
Stats 1418.8: gcodein=67 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1419.8: gcodein=67 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1420.8: gcodein=67 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1421.8: gcodein=67 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1422.8: gcodein=72 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1423.8: gcodein=72 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1424.8: gcodein=72 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1425.8: gcodein=72 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1426.8: gcodein=72 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1427.8: gcodein=77 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1428.8: gcodein=77 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1429.8: gcodein=77 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1430.8: gcodein=77 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1431.8: gcodein=77 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1432.8: gcodein=82 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1433.8: gcodein=82 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1434.8: gcodein=82 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1435.8: gcodein=82 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1436.8: gcodein=82 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1437.8: gcodein=87 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1438.8: gcodein=87 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1439.8: gcodein=87 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1440.8: gcodein=87 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1441.8: gcodein=87 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$
Stats 1442.8: gcodein=92 mcu: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_st$

3DSTB, 2018-03-12:
config-pin P9.41 hi+ has helped

before was 
P9_41 Mode: gpio_pd Direction: out Value: 0

now
P9_41 Mode: gpio_pd Direction: out Value: 1

Looks like i have to figure out that for every pin

closed 2018-03-12 by 3DSTB

Rivega42, 2018-06-02:
Good afternoon. I have a similar problem on ramps1.4 ...
I connect the BL Touch with the instructions "https://github.com/KevinOConnor/klipper/blob/master/config/sample-bltouch.cfg"

"Recv: // Unable to translate pin name: config_end_stop oid = 14 pin = z_virtual_endstop pull_up = 1 stepper_count = 1"

Tell me please how to understand?