Klipper3d/klipper #232: Cramps temperature values not correct

3DSTB, 2018-03-12:
This was already discussed here:
https://github.com/KevinOConnor/klipper/issues/37

I tested also the "Linux host process" with no other result.

I have the same problem. 
May it has somesthing to do with the hardware itself. I hav not much knowledge about that but for me the design looks different than eg from the Ramps:

[image]<br /><a target='_blank' href='https://de.imgbb.com/'>image uploader</a><br />

KevinOConnor, 2018-03-12:
This link: https://groups.google.com/forum/#!topic/machinekit/wZ8KAKqV7yo indicates that the CRAMPS board effectively uses a 2K pullup resistor.  So, add a pullup_resistor=2000 setting to your extruder and heater_bed config sections.

Separately, please choose one issue for all of your CRAMPS issues.  Please do not open new issues - we can track your status much better if you update status in one place.

3DSTB, 2018-03-12:
Thank you. I will test it.

If i have more issues i will go on here: https://github.com/KevinOConnor/klipper/issues/224

3DSTB, 2018-03-12:
The values are correct now.

closed 2018-03-12 by 3DSTB