Klipper3d/klipper #233: No way to support development

mcmasterp, 2018-03-12:
I'm not technically proficient enough to offer direct help with such an advanced project but I do a lot of 3d printing and I am extremely interested in this project advancing. Is there a patreon I can donate to?

theopensourcerer, 2018-03-13:
FWIW, you don't have to be technical to help a project such as this, in fact in some cases it can be very useful; especially with the documentation side.

For example, if there are parts of the notes in the example config files, or installation docs or whatever, that don't make sense it can be helpful to have someone _not_ intimately tied to the code to make the explanations easier to read ;-)

I would think that suggestions to improve stuff like this would usually always be welcome and it will help every user - just as much as new code or bug fixes.

Also, just reporting bugs or issues, although it might not feel very helpful to you, is actually a good contribution to improving the project as a whole.

Forgive me if I am teaching you to suck eggs.

KevinOConnor, 2018-03-14:
On Mon, Mar 12, 2018 at 05:28:52PM +0000, Paul McMaster wrote:
> I'm not technically proficient enough to offer direct help with such an advanced project but I do a lot of 3d printing and I am extremely interested in this project advancing. Is there a patreon I can donate to? 

Thanks.  I'm not personally setup for anything like that right now.  I
will let you know if I do.

Cheers,
-Kevin

KevinOConnor, 2018-03-21:
FYI, I decided to try Patreon: https://www.patreon.com/koconnor

Thanks,
-Kevin

mcmasterp, 2018-03-21:
Great! I signed up. You should create patron levels asap, and write a quick
first post about something like next features your planning and maybe offer
certain level patrons to suggest or vote on possible ideas, like foosel
does for octoprint as an example. And then share this baby everywhere,
groups, forums, at the top of all your web pages, in your github Readme,
everywhere.

closed 2018-04-03 by KevinOConnor

herrwusel, 2018-04-11:
@KevinOConnor: I don't like patreon at all (fee changes and such). Is it a lot work for you to set up a liberapay account as well? I'd like to support you as well :)

KevinOConnor, 2018-04-11:
On Wed, Apr 11, 2018 at 04:45:26AM +0000, herrwusel wrote:
> @KevinOConnor: I don't like patreon at all (fee changes and such). Is it a lot work for you to set up a liberapay account as well? I'd like to support you as well :)

Thanks!  I'm not currently setup for other systems - if I do set
something else up I'll let you know.

-Kevin