Klipper3d/klipper #234: Servo/BLTouch support for Replicape

jakep82, 2018-03-12:
A request to add servo support for the Replicape as referenced in issue #183 .

KevinOConnor, 2018-03-12:
On Mon, Mar 12, 2018 at 10:03:13PM +0000, jakep82 wrote:
> A request to add servo support for the Replicape as referenced in issue #183 .

Can you try the work-replicape-20180312 branch?  Be sure to set the
servo0_enable (and/or servo1_enable) in the config (as described in
the updated config/generic-replicape.cfg file.

cd ~/klipper ; git fetch ; git checkout origin/work-replicape-20180312 ; sudo service klipper restart

-Kevin

jakep82, 2018-03-14:
It's entirely possible I don't have my printer.cfg set up correctly, but I couldn't get this working tonight.  I've attached my log where you can see I played around with a few settings, but nothing would make the BLTouch extend or retract.
[klippy.log](https://github.com/KevinOConnor/klipper/files/1809691/klippy.log)

KevinOConnor, 2018-03-14:
On Wed, Mar 14, 2018 at 05:13:54AM +0000, jakep82 wrote:
> It's entirely possible I don't have my printer.cfg set up correctly, but I couldn't get this working tonight.  I've attached my log where you can see I played around with a few settings, but nothing would make the BLTouch extend or retract.

Okay.  I suspect the klipper code is okay.  What does the following
report?

config-pin -q P9_11
config-pin -q P9_14
config-pin -q P9_28
config-pin -q P9_16
config-pin -q P9_25

You may need to run "sudo config-pin overlay cape-universala" first.
(Or, if you're on a new version of the OS, you may need to modify
/boot/uEnv.txt to somehow enable the "universala" cape.)

-Kevin

KevinOConnor, 2018-03-17:
FYI - I went ahead and committed the servo enable code to the master branch (commit e78377ee) as I don't think it's a regression.

closed 2018-03-26 by KevinOConnor