Klipper3d/klipper #235: Klipper /Octoprint stopped my CR-10s Blank screen

7s7, 2018-03-13:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/1805604/klippy.zip)

Hi,
I just installed klipper and all seems to be going Ok but when I tried to print Octoprint (using octopi and CR-10s) said I was printing but the temps of both the Extruder and Hotbed stayed at zero. I decided to do a restart of Octopi through the menu and when I did the screen on my CR-10s went Blank (just the backlight was working). I managed to redownload the marlin 1.1.8 firmware that I was using and it came back to life.
Tried it a second time and the same thing occured except this time I cant bring it back to life, the computer knows there is something attached but Arduino  starts to compile and then fails saying there is nothing connected.
I have been using this setup for months it is only since I read about Klipper and though it looked a brilliant idea that my problems began.

Are you please able to help.

Thanks

Steve

7s7, 2018-03-13:
After deleting usb devices and usb hubs through windows device manager I was able to re-flash the firmware to my CR-10s.
Unfortunately I think I have to re-image the raspberry pi and just put OctoPi on it.
Hoping to hear if there is a solution to my problem as I would really like to install Klipper.
Thanks 
Steve

grantp69, 2018-03-13:
It doesn't sound like Klipper was the problem if after deleting usb devices on your computer allowed you to upload your old firmware. It's not unusual for a Pi image to get corrupted, some sdcards are just not very good. Make sure it's a good quality sdcard, resetup Klipper and see how it goes.

7s7, 2018-03-13:
I have only reflashed the firmware on the printer not touched the octopi, i dont feel that the SD card in the Raspberry PI is the problem as i have not had any other issues. I will go and buy a new SD card anyway and try again.

7s7, 2018-03-14:
Hi,
I have installed Octopi and Klipper on a new SD card and at this stage when I reboot Octopi I am not getting the issue with the Printer losing its firmware.
Thanks
I still getting the issue though where I send a print command to the printer and the fans speed up but the heater bed nor the extruder heat at all the temps stay at 0.
I am using the correct cfg for the CR-10s (printer-creality-cr10s-2017.cfg) running marlin 1.1.8
Any clues
[klippy.zip](https://github.com/KevinOConnor/klipper/files/1809659/klippy.zip)

Steve

Arksine, 2018-03-15:
The current release of Klipper doesn't show any output on the display.  I believe the upcoming release will.  A blank screen does not indicate corrupt firmware on the current version.

I'm confused, are you using Klipper or Marlin?  You can't run Marlin's firmware on the MCU and simultaneously use Klipper.  You need to flash Klipper's firmware.

Make sure you read the installation docs carefully.

theopensourcerer, 2018-03-15:
@Arksine Just FYI, Klipper does output on my display just fine. Am running a pull from master from last week.

Arksine, 2018-03-15:
That is interesting.  I just installed Klipper on an Ender 2 with a tweaked CR10 config.  Based on Kevin's latest issue I had assumed display support hadn't yet been merged.  My screen is blank, but everything else works fine.

Just looked at TH3D's marlin and it appears the Ender 2's display may behave a bit differently from the CR10, I'll have to look into correcting it.  I apologize 7s7, you may indeed have a problem if no display shows up after flashing Klipper on the MCU.

7s7, 2018-03-15:
Thanks for the reply,
When it came up with no display I did not go any further with testing Klipper.
When I get home this afternoon I will reinstall (third time) and ignore any display issues.
I will then test Klipper and see what happens.

Steve

7s7, 2018-03-16:
Hi Arksine,
Re-installed Klipper and actually tried to run it knowing the display might not work and its works. (Very Impressed
I think the bits that confused me was in the install notes it didn't actually say to reboot the printer, so when I flashed the firmware it still showed the Marlin screen (dumb of me I know I assumed that Klipper ran on top of the main firmware). Its then when i rebooted that the screen went blank (notes said there was basic display support) that I thought I was in trouble.
Thanks for the Help looking forward to you being able to find the display issue.
Let me know if I can be of any help.

Steve

KevinOConnor, 2018-03-17:
Great.  There is basic display support, but it must be configured in the printer.cfg file for it to be enabled.  Take a look at the "display" config section in config/example-extras.cfg, or if you have a ramps board, take a look at the config/generic-ramps.cfg file.

-Kevin

7s7, 2018-03-19:
Hi Kevin,

Thanks for the reply, 
Used the # "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
[display]
lcd_type: st7920
cs_pin: ar16
sclk_pin: ar23
Works great, loving your firmware, 
I can sit for hours watching it, very smooth, beautiful to watch. 
Do you have a PayPal or donation page

Thanks
Steve

KevinOConnor, 2018-03-21:
Glad to hear that Steve!  FYI, I'm in the process of setting up a Patreon page: https://www.patreon.com/koconnor

Thanks,
-Kevin

jfrilot, 2018-03-21:
@KevinOConnor are you planning to add working display settings to configs to help new users?

KevinOConnor, 2018-03-22:
On Wed, Mar 21, 2018 at 06:35:54PM +0000, jfrilot wrote:
> @KevinOConnor are you planning to add working display settings to configs to help new users?

In general, I would like to see the example configs show the
appropriate settings for common displays.  I've added settings to a
few configs (rambo and ramps).  However, I'm relying on others (that
actually have the boards) to submit "pull requests" for the other
configs.

-Kevin

closed 2018-03-26 by KevinOConnor

ChiliApple, 2018-04-22:
Hi,

is it possible to get the "%" display in the ST7920 like this here?
https://www.bountysource.com/issues/44223336-physical-ui-support
the others display works well in the screen on my CR10S

Thx

KevinOConnor, 2018-04-23:
On Sun, Apr 22, 2018 at 03:44:28PM +0000, ChiliApple wrote:
> is it possible to get the "%" display in the ST7920 like this here?
> https://www.bountysource.com/issues/44223336-physical-ui-support
> the others display works well in the screen on my CR10S

If you mean the print progress display, then you need to define a
[virtual_sdcard] section (see config/example-extras.cfg).  The
progress will currently only update if you print from virtual sdcard.

-Kevin

ChiliApple, 2018-04-23:
Hi Kevin,

thanks so much for your answer! i will try it.

Best, Udo

ChiliApple, 2018-04-23:
I still have a question about the history. I now have the progress indicator in the display but it remains at "0" although I print from the virtual SD.
the timer will always display incorrectly, occasionally going back to "0" during printing.

Thx, best Udo