Klipper3d/klipper #238: Planning of v0.6.0 release

KevinOConnor, 2018-03-13:
This issue is just a notice of my intent to make the next Klipper release (v0.6.0). My
rough plan is to make the release around the end of the month (~20180331), with
the first release candidate at the end of this week (~20180318).

If anyone is actively developing a new feature and would like it to be
included in the next release, please let me know this week.  After the
first release candidate, the Klipper master branch will focus on bug
fixes and documentation updates.

The following is my draft list of new features added since the last
release (v0.5.0):
- Initial support for checking of heater and thermistor hardware failures
- Support for Z probes
- Initial support for automatic parameter calibration on deltas (via a
  new delta_calibrate command)
- Initial support for bed tilt compensation (via bed_tilt_calibrate
  command)
- Initial support for "safe homing" and homing overrides
- Initial support for displaying status on RepRapDiscount style 2004
  and 12864 displays
- New multi-extruder improvements:
  - Support for shared heaters
  - Initial support for dual carriage setups
- Support for configuring multiple steppers per axis (eg, dual Z setups)
- Support for custom digital and pwm output pins (with a new SET_PIN command)
- Initial support for a "virtual sdcard" that allows printing directly
  from Klipper (helps on machines too slow to run OctoPrint well)
- Support for setting different arm lengths on each tower of a delta
- Support for G-Code M220/M221 commands (speed factor override /
  extrude factor override)
- Several documentation updates:
  - Many new example config files for common off-the-shelf printers
  - New multiple MCU config example
  - New bltouch sensor config example
  - New FAQ, config check, and G-Code documents
- Initial support for continuous integration testing on all github commits
- Several bug fixes and code cleanups

-Kevin

daboi, 2018-03-13:
Hi Kevin, I am really looking forward to try Klipper soon but I use EZABL for autoleveling. Does this Z probe feature will support ABL sensors?

PRouzeau, 2018-03-15:
Hello,
Does the heater and thermistor failure detection includes 'thermal runaways' ?
And on a more general safety viewpoint, how does Klipper stand on a general safety list as this one : http://reprap.org/wiki/Board_safety

KevinOConnor, 2018-03-17:
FYI, as part of the lead up to the v0.6.0 release I have made some non-backwards compatible config file changes.  Those using static_pwm_output, pwm_output, or digital_output config sections will now need to change their config to use an output_pwm config section.  Those using a probe config section will now need to specify the z_offset as part of that config section.  See the config/example-extras.cfg file for details

decor82, 2018-03-17:
Will TMC2130 with SPI supported in v0.6.0?

alekssobolewski, 2018-03-18:
@ decor82 I'd love to see it there too!
Now I'm using _work-tmc2130-20180217_ branch and I have a problem with current setting. Motors are very noisy! @KevinOConnor Are you planning to create a better library for TMC2130?
I also tried @cruwaller fork, but unfortunately I didn't manage to make it working correctly due to bed heater error :/ (PID Autotune also doesn't work in his fork)
However, the TMC library itself looks very good (as they are in StealchChop).

[printer.txt](https://github.com/KevinOConnor/klipper/files/1822596/printer.txt)

KevinOConnor, 2018-03-18:
@decor82 / @alekssobolewski - Alas, tmc2130 support will not be in v0.6.0.  Best place to track tmc2130 support is issue #101.

KevinOConnor, 2018-03-18:
FYI, I have made another non-backwards compatible change as part of the lead up to the v0.6.0 release.  Support for the M303 g-code command has been removed and a new PID_CALIBRATE command has been introduced.  The Config_checks.md document has also been updated to recommend running the PID_CALIBRATE command.

alekssobolewski, 2018-03-18:
@KevinOConnor I have red the linked issue a few times, but I still have problem. When I try to heat the bed I got an temperature error ant everything stops. Can you recommend a relatively stable fork that supports TMC2130? Are you going to support TMC2130 in Klipper at all?

tiziano149, 2018-03-19:
I am not sure if this is in development or not, but I think including live adjustment of Pressure advance with GCode (e.g. Marlin: M900) would be a much appreciated feature, as it allows you to easily set the best value during calibration and also include a command in the starting script of each filament profile in your slicer, without the need to go into the config file every time you decide to change material.

KevinOConnor, 2018-03-19:
On Sun, Mar 18, 2018 at 09:56:57AM -0700, alekssobolewski wrote:
> @KevinOConnor I have red the linked issue a few times, but I still have problem. When I try to heat the bed I got an temperature error ant everything stops. Can you recommend a relatively stable fork that supports TMC2130? Are you going to support TMC2130 in Klipper at all?

I'd like to see TMC2130 support added, but I don't have any timeframe
for that.  (Other than that it wont make v0.6.0.)  I'd prefer to keep
this issue focused on the v0.6.0 release, so best would be to continue
with the discussion on issue #101.

-Kevin

kevinrocksman, 2018-03-20:
ABL (Z probe) support would be awesome! Cannot wait!

KevinOConnor, 2018-03-31:
FYI, the v0.6.0 release is now available.  Final release notes are at:

https://github.com/KevinOConnor/klipper/blob/master/docs/Releases.md

closed 2018-04-11 by KevinOConnor