Klipper3d/klipper #240: Delta calibration when bed is lower than Z=0 (feature request?)

sapell, 2018-03-14:
Hi,

I greatly appreciate the addition of DELTA_CALIBRATE command. However, is it possible to even temporarily allow Klipper to move the nozzle below Z=0, other than increasing the position_endstop values in printer.cfg? It'd be very convenient if Klipper would automatically allow negative Z position when running DELTA_CALIBRATE.

KevinOConnor, 2018-03-14:
On Wed, Mar 14, 2018 at 08:42:59PM +0000, Simo Apell wrote:
> I greatly appreciate the addition of DELTA_CALIBRATE command. However, is it possible to even temporarily allow Klipper to move the nozzle below Z=0, other than increasing the position_endstop [...]

This can be done with the minimum_z_position setting in the [printer]
config section.

-Kevin

sapell, 2018-03-14:
Thanks Kevin for the quick response! That parameter has not caught my eye for some reason, but thank you for the tip!

closed 2018-03-15 by sapell