Klipper3d/klipper #241: problem with the temperature

Philjayhill, 2018-03-15:
Hi everyone, I have had a problem with the temperature of the 3D printer for days. I've been using the Klipper software for printing since January. Everything works great together with octopi but now the print stops before it reaches the target temperature. has anyone had that?

[klippy.log](https://github.com/KevinOConnor/klipper/files/1816364/klippy.log)

KevinOConnor, 2018-03-15:
On Thu, Mar 15, 2018 at 05:05:34PM +0000, Philjayhill wrote:
> Hi everyone, I have had a problem with the temperature of the 3D printer for days. I've been using the Klipper software for printing since January. Everything works great together with octopi but now the print stops before it reaches the target temperature. has anyone had that?

Your extruder is warming very slow and it's triggering the new minimum
heating rate checks.  The latest version of the code improves the
error message a little.  If you want to continue with that slow
heating, you'll need to add a "[verify_heater extruder]" section to
your config and modify the values for the heating_gain and/or
check_gain_time fields.  See config/example-extras.cfg for further
detail.

-Kevin

Philjayhill, 2018-03-16:
Hello Kevin, first of all thank you for your post.
I've made these changes now, but it doesn't seem to help.
[verify_heater extruder]
heating_gain: 2
check_gain_time:35
hysteresis: 10
max_error: 130
I also changed the whole hotend because I thought it might help. I'm getting desperate because I really need to get the printer up and running.

Here is the logfile:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1818696/klippy.log)

KevinOConnor, 2018-03-16:
On Fri, Mar 16, 2018 at 02:44:31AM -0700, Philjayhill wrote:
> Hello Kevin, first of all thank you for your post.
> I've made these changes now, but it doesn't seem to help.
> [verify_heater extruder]
> heating_gain: 2
> check_gain_time:35
> hysteresis: 10
> max_error: 130

The log indicates that a target temperature of 195 degrees was set,
but even with the extruder heater fully enabled the temperature
plateaued at 169 degrees.

That's highly unusual, and you should carefully check if there is
something wrong with the printer.

If you normally wait a half an hour or so to get your extruder up to
the target temperature, and you really want to continue, then you can
completely disable the heating rate check by setting check_gain_time
to 999999999999.

-Kevin

Philjayhill, 2018-03-16:
i checked everything a few times and also installed new parts in the printer, i don't think the printer has a hardware problem. My next attempt after setting the check_gain_time to 99999999999999 the printer after reaching the target temperature simply switched off after a few seconds. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/1819383/klippy.log)

Philjayhill, 2018-03-16:
i checked all plugs and connections again... which i find funny, if i just set the temperature to 195°, so preheat it goes very fast until it is at the desired temperature, but as soon as i want to print something it goes quite long until the hotend has reached the temperature

KevinOConnor, 2018-03-16:
On Fri, Mar 16, 2018 at 01:58:47PM +0000, Philjayhill wrote:
> i checked all plugs and connections again... which i find funny, if i just set the temperature to 195°, so preheat it goes very fast until it is at the desired temperature, but as soon as i want to print something it goes quite long until the hotend has reached the temperature

As soon as you turned on your bed, the extruder started to rapidly
cool.  My guess is your power supply isn't powerful enough to drive
both heaters.  I advise caution here - driving your electronics that
close to the edge could cause damage.

Some things that you may want to explore:

- set the heater_bed max_power to 0.250 so it isn't left fully on for
  extended periods (this is assuming your heater circuit is okay with
  switching at a 100ms cycle rate)
- run a pid tune on your bed (M303 E-1 S65) and use pid control on the
  bed (again, assuming the heater circuit is okay with pwm switching)
- run a pid tune on your extruder (M303 E0 S190) and use the
  recommended settings
- revert the verify_heater changes to your config

If, instead, you want to go forward without the verify_heater checks,
then you can set the max_error field also to 9999999999 to disable
that check as well.  (I don't recommend this.)

-Kevin

closed 2018-03-26 by KevinOConnor

bestsase, 2019-02-07:
Hello, i have similar problem but only when i set extruder heater to heat while its cooling down from previous print, as example i was printing with ABS, it was short test print and it finished, extruder was set to 240c, it cooled down to ~190 and i set it again to 240 manually to prepare it for new print, then it throws error:
Recv: // Heater extruder not heating at expected rate 
Recv: // See the 'verify_heater' section in config/example-extras.cfg 
Recv: // for the parameters that control this check.

now i added check interval to 120sec (instead of default 20sec) to see what will happen...
[verify_heater extruder]
check_gain_time: 120

Am i doing something wrong?

RoamingInterests, 2021-10-13:
Why is this feature enabled for? And how can it be changed or disabled so it doesn't stop the prints? 
How can i disable this temperature problem?