Klipper3d/klipper #246: PD_CALIBRATE missing from latest master branch

w1ebr, 2018-03-18:
Hi Kevin,

I know you moved PID_CALIBRATE to its' own file. When I issue the command in Octoprint, it returns

Send: PID_CALIBRATE HEATER=extruder TARGET=248
Recv: // Unknown command:"PID_CALIBRATE".

Is there an include missing? (guess)

Thanks!

Gene

KevinOConnor, 2018-03-18:
On Sun, Mar 18, 2018 at 02:50:08PM -0700, w1ebr wrote:
> Hi Kevin,
> 
> I know you moved PID_CALIBRATE to its' own file. When I issue the command in Octoprint, it returns
> 
> Send: PID_CALIBRATE HEATER=extruder TARGET=248
> Recv: // Unknown command:"PID_CALIBRATE".
> 
> Is there an include missing? (guess)

It works fine for me.  Can you attach the klippy.log file?

-Kevin

w1ebr, 2018-03-18:
Hi Kevin, I changed the title when I realized what had happened. The real issue is related to the new title :-)

Thanks!

Gene

KevinOConnor, 2018-03-19:
Klipper should be extended with the ability to add custom thermistor definitions via the config file.  I don't see that feature being added before the v0.6.0 release though.

Separately, the Linear temperature conversion code only looks at t1/v1 and t2/v2 - the other values in your snippet above would be ignored.

-Kevin

w1ebr, 2018-03-19:
Thanks, Kevin!

KevinOConnor, 2018-04-09:
This should be possible now as of commit 6466af63.  Both thermistors and "linear" sensors can now be defined in the config file.  Also, there is a new "PT100 INA826" sensor type defined by default now (which also supports a range of sensor values, so should be more accurate at the temperature extremes).

-Kevin

closed 2018-04-18 by KevinOConnor