Klipper3d/klipper #247: Feature Request: Pressure Advance gcode.

tiziano149, 2018-03-19:
I am not sure if this is in development or not, but I think including live adjustment of Pressure advance with GCode (e.g. Marlin: M900) would be a much appreciated feature, as it allows you to easily set the best value during calibration and also include a command in the starting script of each filament profile in your slicer, without the need to go into the config file every time you decide to change material.

KevinOConnor, 2018-03-19:
FYI, this was raised previously in issue #92.

ml0w, 2018-03-20:
Some good suggestions raised in #92 why close? Maybe I'm confused how github issues work. What needs to be done to implement functionally the same effect as M900 Kn? Options to set config thru command? All of my profiles in slic3r send M900 at start of print with value for material type, which with a long bowden can vary by factor of 2x. Having to restart klipper and manually ssh in and change a file seems a bit much when I change from petg to pla/etc.

KevinOConnor, 2018-03-22:
On Tue, Mar 20, 2018 at 03:33:58AM -0700, ml0w wrote:
> Some good suggestions raised in #92 why close? Maybe I'm confused how github issues work. What needs to be done to implement functionally the same effect as M900 Kn? Options to set config thru [...]

I prefer to keep the open issue list to active items.  I fear the list
of open issues becomes hard to manage when there are many feature
requests open that aren't being worked on.

I don't see M900 being directly supported, but it would be possible to
add a "SET_PRESSURE_ADVANCE advance=0.0123" command.  That said, it's
unclear to me whether we'd want to do that or instead add a
"SET_PROFILE profile=my_petg_profile" type command.  In either case,
this would be something to look at after the v0.6.0 release.

-Kevin

tiziano149, 2018-03-23:
I see how setting profiles via command could be useful.

The real benefit in changing individual values like pressure advance, junction deviation or acceleration with g-code is live tuning, as it enables to finetune these settings as you print a test object. Saves a lot of guesswork and quickly shows you how changing settings affects print speed and quality. An essential feature in my view.

bmc0, 2018-04-01:
Here's a quick-and-dirty patch that allows setting `pressure_advance` and `pressure_advance_lookahead_time`:

[extruder-Add-SET_PRESSURE_ADVANCE-command.patch.txt](https://github.com/KevinOConnor/klipper/files/1866653/extruder-Add-SET_PRESSURE_ADVANCE-command.patch.txt)

Usage: `SET_PRESSURE_ADVANCE [ADVANCE=<pressure_advance>] [ADVANCE_LOOKAHEAD_TIME=<pressure_advance_lookahead_time>]`

KevinOConnor, 2018-04-03:
On Sun, Apr 01, 2018 at 01:01:06PM -0700, Michael Barbour wrote:
> Here's a quick-and-dirty patch that allows setting `pressure_advance` and `pressure_advance_lookahead_time`: [...]

Thanks.  If you'd like me to commit this, can you include a
signed-off-by line (as described in
https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md
).

One thing the patch is missing is flushing of the lookahead queue.
This can be done with a call to toolhead.get_last_move_time() at the
top of cmd_SET_PRESSURE_ADVANCE().  (If the queue isn't flushed, I
fear the internal lookahead logic could get out of sync and lead to
weird errors.)

-Kevin

bmc0, 2018-04-03:
Ah, thanks. Figured I may have missed something. I submitted a pull request with a few changes.

KevinOConnor, 2018-04-03:
Thanks @bmc0. This was applied to the master branch as of commit a3017133.

closed 2018-04-09 by KevinOConnor