Klipper3d/klipper #249: Request to use ilamnet Run Out Sensor

MarcinKnD, 2018-03-21:
Can you add RunOut sensor? its connected to Servo1 (D6) on Ramps, I like this from Repetier Firmware

leungtech, 2018-03-25:
yes, it is very important, please add,thx!!!

Duckle29, 2018-04-03:
I guess this could be thought of as a request to implement M600 as done in marlin / smoothieware: http://reprap.org/wiki/Gcodes#M600:_Filament_change_pause

MarcinKnD, 2018-04-20:
M600 It Will be great.

MarcinKnD, 2018-05-19:
OK. I Have connect the Runout sensor to Orange PI and its finaly works, thaks to arhi (Bogdan Kecman) and his Octoprint-Filament-Reloaded-OrangePi.

Now the sensor Trigger the pause, 
and do GCODE
 After pause:
G91
G1 Z50
G1 E-40 F500
M104 S0
M140 S0

I change the filamnet

and after resume do gcode
G91
G1 Z50
G1 E-40 F500
M104 S0
M140 S0

When printer start printing its give 
> Your printer's firmware reported an error. Due to that the ongoing print job will be cancelled. Reported error: Move exceeds maximum extrusion (4.937mm^2 vs 0.640mm^2)

If we can after triger to pause remember all position XYZE, ten switch to relative, make some moves home XY, park in position to change fimalent, retract 4mm material, then after resume extract 3-4 mm material and then go to XYZE where pause whas trigered, than switch to abloute shold do the trick.

ChiliApple, 2018-05-28:
it only works when you print from local space, not from virtuell SD!

KevinOConnor, 2018-07-07:
With the addition of "buttons" support (commit 38d7b9ad) it should now be possible to put together a klipper host python "module" to support filament runout sensors.  I don't have one of these sensors, but if someone with the sensor is up for doing some development work, it should be possible.

-Kevin

closed 2018-07-17 by KevinOConnor

mamoit, 2018-08-07:
An example in the `example-extras.cfg` would be perfect. I'm trying to figure out how to do it and can't seem to find a way.

mamoit, 2018-08-07:
Something like [this](https://github.com/mamoit/klipper/tree/PR-runnout)?

KevinOConnor, 2018-08-16:
@mamoit - That code looks like a valid way to use the buttons module.  Please submit a pull request if you would like me to do a full review and you want to include it in mainline.

-Kevin

mamoit, 2018-08-17:
PR created #545 still needs a ton of work, but I want your input.

KevinOConnor, 2019-05-26:
Basic filament sensor support is now in Klipper.

-Kevin

PR #3857 (2021-03-15): Add filament_motion_sensor
Add functionality to support a Filament Motion Sensor for detecting extruder jams as well as runouts.  Works by an encoder toggling the switch_pin 0/1 as the filament is pulled through the sensor.

Everything seems to work ok, except I have seen some weirdness in pause_resume.  When pausing, the printer runs the runout_gcode however it sometimes moves back over the print job (does not affect Z, only X (and maybe Y)).  Increasing the pause_delay may resolve this.

Also pause_resume seemed a little inconsistent in how it was handled so I tried to address, not sure if t [... cut]