Klipper3d/klipper #251: disable end stops

ejg3855, 2018-03-22:
I have everything up and running, I don't have all the hardware sorted out yet. I am using leadscrew designs based on what we do at work (linear motion company), but I don't want to commit to building the rest of the printer if I cannot power the steppers. I have concern that the DRV8825 just don't have enough current capacity to reliably move what I want to do (boss who knows more about steppers than anyone I've met or read online said its borderline) so I am trying to get a real barebones setup so I can just try to drive the Y-axis. 

I have commented out the X/Z axis but get errors that say they must exist, I tried commenting out the end stop pins but also get errors. I tried to use the marlin style M121 command to disable endstops but klipper doesn't recognize this. 

I really just want to move my y-axis so I can test my hardware. is this possible? does this need a log?

ejg3855, 2018-03-22:
I just keep getting: Recv: !! Must home axis first: 0.000 10.000 0.000 [0.000]

hg42, 2018-03-22:
You can keep all axes configured and simply not connect anything to X and Z pins (remove motors).

Homing is a problem. Kevin thinks it's too dangerous to move the toolhead if klipper doesn't know the position.
I don't know how he currently handles such things.

I once offered my own patch for this, but now I use cruwaller's fork. He has a setting to allow_move_without_home.
If you want to use Kevin's repo, then you could patch it yourself.

It's loaded in toolhead.py:
```
        self.allow_move_wo_homing = config.getboolean(
            'allow_move_without_home', False)
```

and used in kinematic code (either cartesian or corexy or delta), e.g. in cartesian.py:
```
        if toolhead.allow_move_wo_homing is False:
            self.limits = [(1.0, -1.0)] * 3
        else:
            # Just set min and max values for SW limit
            self.limits = [ (s.position_min, s.position_max)
                            for s in self.steppers ]
```

KevinOConnor, 2018-03-22:
On Thu, Mar 22, 2018 at 12:25:38AM +0000, ejg3855 wrote:
> I have everything up and running, I don't have all the hardware sorted out yet. I am using leadscrew designs based on what we do at work (linear motion company), but I don't want to commit to [...]

The easiest way to do that is to define the X and Z steppers using
pins that are otherwise unused.  (Klipper may toggle these pins at
runtime, but that should be harmless if they are otherwise unused.)

On Thu, Mar 22, 2018 at 12:34:12AM +0000, ejg3855 wrote:
> I just keep getting: Recv: !! Must home axis first: 0.000 10.000 0.000 [0.000]

The easist way to disable homing is to add the following to your
config:

[homing_override]
set_position_x: 0
set_position_y: 0
set_position_z: 0
gcode:

You'll still need to run G28 at startup, but the G28 wont actually
move anything.  See config/example-extras.cfg for further details.

-Kevin

ejg3855, 2018-03-22:
thank you gents.

ejg3855, 2018-03-23:
i ran this code and my y-axis traveled. It wouldn't move in the negative Y direction G1 Y-5 example but I proved I could move without trouble,

thanks!

closed 2018-03-26 by KevinOConnor