Klipper3d/klipper #252: Request Endstop distance after homing

MarcinKnD, 2018-03-22:
Can You add Endstop distance after homing. Now after homing the axis is constanty triger the endstops. It will be nice to bump 2mm afer homing

cmock, 2018-03-23:
I would use the ``[homing_override]`` section, see example_extras.cfg, and then put something like

     G28
     G1 X2 Y2 Z2

in that section. (And I'd be sure to have my finger on the power switch when first using this, because I haven't tested it).

MarcinKnD, 2018-03-23:
thanks a lot it wokrs.

I  modify a little bit the Gcode, for my CoreXY printer, I whant to Home first Z Axis, then te Y and at the end X axis
here is the code:
`[homing_override]
gcode:
	G28 Z0
	G1 Z295
	G28 Y0
	G1 Y2
	G28 X0
	G1 X2
`

Thanks

closed 2018-03-31 by KevinOConnor