Klipper3d/klipper #254: Feature Idea/Suggestion: Octoprint printer.cfg editor

theopensourcerer, 2018-03-23:
Just a thought. 

I know on the RepRap Firmware you can edit the config file from within the web browser. I thought this might be a nice feature for the future? (Maybe with an auto "reload" on save?)

I've not build any plugins for Octoprint before but would be happy to have a look into this if others think it might be a good idea...

theopensourcerer, 2018-03-23:
It looks like a great deal of the effort required has already been accomplished:

https://github.com/ieatacid/OctoPrint-GcodeEditor

gdachs, 2018-03-25:
Klipper is totally independent from OctoPrint. So this plugin is of no use for Klipper. Klipper makes it very easy to edit the configuration and it has not to be transferred to the printer board. So what is wrong with just using vi? It is not a daily task to edit the config.

theopensourcerer, 2018-03-26:
@gdachs Thanks for the input.

I know that Klipper and Octoprint are independent but I guess the vast majority of installations do use Octoprint as the web-front end for controlling the printer - so from the user's perspective this is very similar to how RRF works. It seemed to me it would be a "nice" idea to be able to edit the config file without _having_ to use ssh and a cli editor.

Web editing of the config file is the default/standard way for RRF. 

At some point in the future, hopefully, Klipper will be used by people who are not familiar with vi  - let alone emacs ;-) 

For some of us, changing the config file _is_ a frequent task - especially when changing toolheads etc...

theopensourcerer, 2018-03-26:
Thanks @KevinOConnor  - I'll see if I can touch base with @mmone.. 

Don't hold your breath but this is something I'd quite like to try and see if I can do or help with at least.

I'm guessing handling file permissions on the config file might be the awkward thing so Octoprint can write to it. I've not got a Pi in front of me right now but I guess the printer.cfg is 644 (rw-r--r--) klipper and octoprint will run as it's own user... Would you have any objections to making the config file 664 klipper:octoprint at some point - or I could include an instruction on how to do this in the plugin and an explanation of what it means.

KevinOConnor, 2018-03-26:
On Mon, Mar 26, 2018 at 03:52:52PM +0000, Alan Lord wrote:
> I'm guessing handling file permissions on the config file might be the awkward thing so Octoprint can write to it. I've not got a Pi in front of me right now but I guess the printer.cfg is 644 [...]

The default instructions install Klipper in the same user account as
OctoPrint.  I think it's safe to assume they will be the same.

-Kevin

theopensourcerer, 2018-03-26:
Oh - even better. Awesome :-)

mmone, 2018-03-26:
Hi @theopensourcerer,

in my opinion generating a config is a onetime thing that doesn't warrant an editor at the moment. Especially before 1.0 and a fairly fixed interface.

Your usecase to quickly change specific parameters in production might be easier covered by other means. The plugin already allows to add custom buttons. It would be possible to add additional functionality to explicitly write a parameter in the Klipper config with a special "macro" that gets parsed.
Much cleaner though would be to add this functionality on the Klipper side as an "Extra" with two commands "READ_PARAM <name> WRITE_PARAM <name>. 

If you want to give it a try, the octoprint plugin API is quite well document: [docs.octoprint.org](http://docs.octoprint.org/en/master/plugins/index.html) It uses knockout.js plus Bootstrap on the frontend putting together a form is relatively simple.

theopensourcerer, 2018-03-26:
@mmone Thanks for the follow up.

KevinOConnor, 2018-05-14:
I'm going to close this for now.  Happy to reopen if someone is actively working on the development of it.

Cheers,
-Kevin

closed 2018-05-14 by KevinOConnor