Klipper3d/klipper #256: Allow for different max velocity to be for X and Y accesses

mikedahlgren, 2018-03-24:
First thank you for the fantastic work you've done with Klipper!

Having a mostly custom printer loosely based around the CR-10s I "upgraded" the Y access to a ball screw while the X access is still a belt.  The result I found is the real world max speed for the Y access is significantly slower than the X.  It would be fantastic if the max speed of each access could be defined separately e.g. 

max_x_velocity = 50
max_y_velocity = 250

KevinOConnor, 2018-03-26:
On Fri, Mar 23, 2018 at 10:18:41PM -0700, Mike Dahlgren wrote:
> First thank you for the fantastic work you've done with Klipper!
>
> Having a mostly custom printer loosely based around the CR-10s I "upgraded" the Y access to a ball screw while the X access is still [...]

In this situation, I'd recommend you set the max_velocity to the lower
of the two (ie, max_velocity=50).  If one were to program Klipper with
different X and Y maximum velocities, then in practice just about
every move would have a different velocity (relative to the print).
(Each diagonal move will use a slightly different amount of X movement
vs Y movement, and thus the velocity limits will impact each diagonal
differently.)  That will likely cause non-optimal results due to the
constant changes in extruder flow rate.

If you wish to play with it, take a look at
klippy/cartesian.py:check_move() - the code to slow down the X would
be similar to the existing code there that slows down the Z.

Given the concerns though, I don't think this type of setting makes
sense to put in the main code.

-Kevin

closed 2018-04-03 by KevinOConnor

dpellegr, 2021-11-28:
Hi, I am trying to gain a better understanding of this issue.

In the OP case, the two axes have different rotation distances (or gearing). So the low-geared motor should be spinning fast, while the speed of the other motor should be adjusted so that the two axes have the same linear speed. This should be simply accomplished with the `rotation_distance` (or `gearing`) setting, am I wrong?

Then we come to the case where the two axes have different inertia, and therefore could have different max linear speed. Clearly now the maximum velocity of the printing head will have a strong dependency on the direction. But isn't this already the case with a cartesian frame? 45 degrees moves where both axes operate at their max speed are sqrt(2) times faster than pure X or Y moves. So unless we limit the total velocity instead of its projections, we need to adjust the extruder flow with the velocity direction anyway, no?

Here are a few plots showing the max print head velocity according to the polar direction in the XY plane:

[image]
[image]
[image]

Clearly the more the X and Y max velocity differs, the larger the velocity swings of the head are, but I don't think that the argument of the extruder flow variation is a show stopper, as we have that problem anyway.

Eventually, for a nice and even flow in vase mode, one should fix the maximum print head velocity from the slicer, making sure that it is compatible with the slower axis.

-----------------------

PS Here is the python code to reproduce the plots above:

```
import matplotlib.pyplot as plt
import numpy as np

Xmax = 1
Ymax = 1

theta = np.arange(0,2*np.pi,0.01)
tan = np.tan(theta)

r = np.sqrt( np.minimum( \
  np.square(Xmax)+np.square(Xmax*tan), \
  np.square(Ymax)+np.square(Ymax/tan)  \
))

plt.title("Xmax = "+str(Xmax)+"    Ymax = "+str(Ymax))
plt.xlabel("Direction")
plt.ylabel("Max Speed")
plt.plot(theta*180/np.pi, r)
plt.show()

```